JeffyCN/rockchip_mirrors #2: can we mirror the source code of xf86-video-armsoc?

amazingfate, 2020-10-16:
@JeffyCN hello~
I have a deb installed on an old rk3288 rootfs image:
```
 dpkg -l|grep armsoc
ii  xserver-xorg-video-armsoc                     1.9.3                             armhf        X.Org X server -- ARM SoC display driver (transitional package)
```
This is the debian changelog of that package:
```
xf86-video-armsoc (1.9.3) testing; urgency=medium

  * new

 -- Jacob Chen <jacob2.chen@rock-chips.com>  Fri, 03 NOV 2017 13:17:50 +0200
```
This version is not compatible with xorg 1.20+, so I want to rebuild this package since armsoc+fbdev+xinerama is the only way I found to light up rk3288 + displaylink usb2.0 multi screen.

JeffyCN, 2020-10-16:
that is not supported(either armsoc driver or mali fbdev).

newest 3288 debian sdk is using modesetting

JeffyCN, 2020-10-16:
If you want to find armsoc sources(not updated for years), maybe try to google rockchip armsoc, for example:
https://github.com/markyzq/armsoc-rockchip <--- a bit newer
https://github.com/mmind/xf86-video-armsoc

closed 2020-10-16 by JeffyCN