JeffyCN/rockchip_mirrors #4: [Request] aarch64 versions of the rkbin/tools binaries

Strit, 2021-07-16:
Right now, it's not possible to compile uboot and create the signed uboot binaries (idbloader.img, idblock.bin and trust.img) on an aarch64 device, like the Quartz64.

So it's not possible to build the software on the device it's suppose to run on.
And that's because all the binaries in rkbin/tools/ are `x86_64`, so they are not able to be executed on `aarch64` based hardware.

I would like to request aarch64 versions of those binaries. They could even just be placed in an rkbin/tools/aarch64 folder or something like that.

Thanks for reading this request.

JeffyCN, 2021-07-16:
Hi,

This is just a mirror repo, so please mail the u-boot maintainer (Joseph Chen <chenjh@rock-chips.com>) for further support.

AFAIK, building the BSP u-boot depends on a few shell scripts and a few pre-built x86 tools. And those tools are open sourced, might you can try to check u-boot/tools/rockchip/

Strit, 2021-07-16:
Ah, so I could in theory build those tools. Thanks.

closed 2021-07-16 by Strit