JeffyCN/rockchip_mirrors #9: 硬件解码

suyulin, 2021-12-08:
能否给一个硬件解码的示例？
用以下 命令 在3326 平台播放 25fps 视频，有花屏现象。
gst-launch-1.0 filesrc location=vui_1.mov ! decodebin ! videoconvert ! videoscale ! autovideosink

JeffyCN, 2021-12-08:
1/ please use english in issues, so other people could check it too.
2/ gst-play-1.0 xxx.mp4

closed 2021-12-08 by JeffyCN

suyulin, 2021-12-09:
hi, Jeffy
I play [this video](https://user-images.githubusercontent.com/6275411/145319168-7a71d720-1608-4b00-9bbf-05d696e31796.mov
) on 3326 , has splash screen。
```
GST_DEBUG=2 gst-play-1.0 -v vui_1.mov
Press 'k' to see a list of keyboard shortcuts.
Now playing /data/vui_1.mov
0:00:00.055302041   912     0x306dcac0 WARN                 basesrc gstbasesrc.c:3583:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.058767041   912     0x306dcac0 WARN                 basesrc gstbasesrc.c:3583:gst_base_src_start_complete:<source> pad not activated yet
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0: ring-buffer-max-size = 0
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0: buffer-size = -1
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0: buffer-duration = -1
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0: use-buffering = false
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0: download = false
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0: uri = file:///data/vui_1.mov
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0: connection-speed = 0
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0: source = "\(GstFileSrc\)\ source"
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:src: caps = video/quicktime, variant=(string)iso
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:src: caps = NULL
0:00:00.108712041   912   0x7fa80429e0 WARN                 qtdemux qtdemux_types.c:233:qtdemux_type_get: unknown QuickTime node type tmcd
0:00:00.108993791   912   0x7fa80429e0 WARN                 qtdemux qtdemux_types.c:233:qtdemux_type_get: unknown QuickTime node type nmhd
0:00:00.109066708   912   0x7fa80429e0 WARN                 qtdemux qtdemux_types.c:233:qtdemux_type_get: unknown QuickTime node type tmcd
0:00:00.109256291   912   0x7fa80429e0 WARN                 qtdemux qtdemux.c:3031:qtdemux_parse_trex:<qtdemux0> failed to find fragment defaults for stream 1
0:00:00.109677750   912   0x7fa80429e0 WARN                 qtdemux qtdemux.c:3031:qtdemux_parse_trex:<qtdemux0> failed to find fragment defaults for stream 2
0:00:00.110122250   912   0x7fa80429e0 WARN                 basesrc gstbasesrc.c:2445:gst_base_src_update_length:<source> processing at or past EOS
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0.GstPad:src_0: caps = video/x-h264, stream-format=(string)avc, alignment=(string)au, level=(string)1.3, profile=(string)high, codec_data=(buffer)0164000dffe100176764000dacd307857e4840000003004000000c83c50a4e01000568eae043cb, width=(int)480, height=(int)154, framerate=(fraction)25/1, pixel-aspect-ratio=(fraction)1/1
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstH264Parse:h264parse0.GstPad:sink: caps = video/x-h264, stream-format=(string)avc, alignment=(string)au, level=(string)1.3, profile=(string)high, codec_data=(buffer)0164000dffe100176764000dacd307857e4840000003004000000c83c50a4e01000568eae043cb, width=(int)480, height=(int)154, framerate=(fraction)25/1, pixel-aspect-ratio=(fraction)1/1
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-buffers = 5
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-time = 0
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-bytes = 2097152
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0.GstMultiQueuePad:sink_0: caps = video/x-h264, stream-format=(string)avc, alignment=(string)au, level=(string)1.3, profile=(string)high, codec_data=(buffer)0164000dffe100176764000dacd307857e4840000003004000000c83c50a4e01000568eae043cb, width=(int)480, height=(int)154, framerate=(fraction)25/1, pixel-aspect-ratio=(fraction)1/1
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-buffers = 5
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-time = 0
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-bytes = 2097152
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-buffers = 5
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-time = 0
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-bytes = 2097152
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)1.3, profile=(string)high, width=(int)480, height=(int)154, framerate=(fraction)25/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
mpi: mpp version: Without VCS info
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
/GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMppVideoDec:mppvideodec0.GstPad:sink: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)1.3, profile=(string)high, width=(int)480, height=(in [... cut ...]
/GstPlayBin:playbin/GstInputSelector:inputselector0.GstSelectorPad:sink_0: tags = taglist, video-codec=(string)"H.264\ \(High\ Profile\)", bitrate=(uint)1057795, encoder=(string)Lavf58.29.100, container-format=(string)"ISO\ MP4/M4A", minimum-bitrate=(uint)340400, maximum-bitrate=(uint)2607200;
/GstPlayBin:playbin/GstInputSelector:inputselector0.GstSelectorPad:sink_0: tags = taglist, video-codec=(string)"H.264\ \(High\ Profile\)", bitrate=(uint)1057795, encoder=(string)Lavf58.29.100, container-format=(string)"ISO\ MP4/M4A", minimum-bitrate=(uint)340400, maximum-bitrate=(uint)2614000;
```
What is the solution, looking forward to your reply

JeffyCN, 2021-12-09:
Hi,

gst mpp plugin is just a wrapper for MPP.

for this kind of corruption, most likely due to MPP(user space lib or kernel driver), and might even related to special chips(it works well on my 3399 evb).

so, please try MPP's test(mpi_dec_test) to confirm, if it's an MPP issue, report to https://github.com/rockchip-linux/mpp/issues and maybe mail herman.chen@rock-chips.com.

1: extract raw h264 stream(PC):
ffmpeg -i 145319168-7a71d720-1608-4b00-9bbf-05d696e31796.mov -map 0:0 -vcodec copy -an -f h264 test.h264
2: try mpi_dec_test(device):
mpi_dec_test -i test.h264 -t 7 -w 480 -h 160 -o test.nv12
3: use 7-yuv or other tools to check the output test.nv12 file(PC)

suyulin, 2021-12-09:
I update `librockchip_mpp.so ` and  `librockchip_vpu.so` this problem has solved.
```
gst-play-1.0 /data/video.h264
Press 'k' to see a list of keyboard shortcuts.
Now playing /data/video.h264
mpp[775]: mpp_info: mpp version: 31814ae author: Herman Chen    2021-09-30 [vepu]: Add offset_x offset_y support for vepu
```

suyulin, 2021-12-14:
hi,
` gst-launch-1.0 -v playbin` is not support hardware decoding