JeffyCN/rockchip_mirrors #12: VP8 encoding results in iommu page fault

andrzejtp, 2022-01-20:
```
mpi_enc_test -i avdec.yuv -o out.enc -w 640 -h 360 -f 4 -t 9 -n 2
```
Frame 0:

```
[   42.213800] rk_iommu ff650800.iommu: Page fault at 0x0000000004382010 of type write
[   42.214509] rk_iommu ff650800.iommu: iova = 0x0000000004382010: dte_index: 0x10 pte_index: 0x382 page_offset: 0x10
[   42.215438] rk_iommu ff650800.iommu: mmu_dte_addr: 0x000000003da7c000 dte@0x000000003da7c040: 0x000000 valid: 0 pte@0x0000000000000000: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[   42.216918] rk-vcodec ff650000.vpu_service: vcodec, fault addr 0x04382010 status 6b
```

```
[   42.248374] rk-vcodec ff650000.vpu_service: reg[44] 04382013
```

Frame 1:

```
[   42.349926] rk_iommu ff650800.iommu: Page fault at 0x0000000000000010 of type write
[   42.350615] rk_iommu ff650800.iommu: iova = 0x0000000000000010: dte_index: 0x0 pte_index: 0x0 page_offset: 0x10
[   42.351503] rk_iommu ff650800.iommu: mmu_dte_addr: 0x000000003da7c000 dte@0x000000003da7c000: 0x000000 valid: 0 pte@0x0000000000000000: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[   42.352966] rk-vcodec ff650000.vpu_service: vcodec, fault addr 0x00000010 status 6b
```

```
[   42.401597] rk-vcodec ff650000.vpu_service: reg[80] 00000010
```

I'm trying this on rk3399. The addresses are set in: `mpp/hal/vpu/vp8e/hal_vp8e_vepu2_v2.c`, `static MPP_RET vp8e_vpu_frame_start(void *hal)`:

```
regs->sw44.base_partition1 =  hw_cfg->partition_Base[0];
```
```
regs->sw80.base_mv_write =  hw_cfg->mv_output_base;
```

and are never mapped by iommu, hence the page fault.

JeffyCN, 2022-01-20:
first of all, this is just a mirror, please report this kind of issues to the real maintainers(or find the mail address of recent commiters)

and for this kind of issues, likely due to mismatched kernel driver and userspace libraries.

closed 2022-01-20 by JeffyCN

JeffyCN, 2022-01-28:
anyway, these kind of issues looks like something wrong in kernel, or at least some version mismatch between kernel mpp driver and userspace mpp library.

you can try the yocto sdk(which should have matched kernel and userspace): https://github.com/JeffyCN/yocto-manifests