JeffyCN/rockchip_mirrors #13: Which rockchip repo should we focus on?

windowsair, 2022-01-28:
Hi, I noticed that the `mpp` and `gstreamer-rockchip` branches do not match the commits of the relevant repo in the [rockchip-linux ](https://github.com/rockchip-linux) organization. 

Does this mean that all future official work is done in this repo? Thank You!

JeffyCN, 2022-01-28:
1/ due to company policy, the rockchip-linux is kind of unmaintained now

2/ there was a few rebases in mpp and gstreamer-rockchip, so the commit id might not matched

3/ the official work is always in rockchip's internal code server, these github repo are just mirrors, i will sync them when needed.

windowsair, 2022-01-28:
Thanks for the quick reply!

I think I may have missed the point, I mean, for the new design (regarding mpp and gstreamer-rockchip), should we start with this mirror repo which looks newer or with the older rockchip-linux organization repo?

JeffyCN, 2022-01-28:
the rockchip-linux repo are unmaintained, so might be out-dated.

my mirrors should always be the newest when i sync it.

the newest gstreamer-rockchip has lots of fixes and new features, so i would suggest using my mirrors.

but new userspace might not compatible with old kernel, so you might need to make sure using the matched kernel

windowsair, 2022-01-28:
make sense. thank you for the clarity!

closed 2022-01-28 by windowsair