JeffyCN/rockchip_mirrors #15: full path when linking with mali library

MastaG, 2022-04-12:
Hi @JeffyCN 

It seems when building the shared objects.
It tends to put the full path to the mali driver instead of just the filename.
```
ldd /usr/local/lib/libgbm.so.1
	linux-vdso.so.1 (0xbed53000)
	libmali_hook.so.1 => /usr/local/lib/libmali_hook.so.1 (0xb6f00000)
	libc.so.6 => /lib/libc.so.6 (0xb6d80000)
	/mnt/sdcard/src/rockchip_mirrors/optimize_3/arm-linux-gnueabihf/libmali-midgard-t62x-r12p0-04rel0-wayland.so (0xb5d00000)
	libdrm.so.2 => /lib/libdrm.so.2 (0xb5cd0000)
	/lib/ld-linux-armhf.so.3 (0xb6f49000)
	librt.so.1 => /lib/librt.so.1 (0xb5cb0000)
	libpthread.so.0 => /lib/libpthread.so.0 (0xb5c90000)
	libdl.so.2 => /lib/libdl.so.2 (0xb5c70000)
	libwayland-client.so.0 => /lib/libwayland-client.so.0 (0xb5c50000)
	libwayland-server.so.0 => /lib/libwayland-server.so.0 (0xb5c30000)
	libstdc++.so.6 => /lib/libstdc++.so.6 (0xb5a50000)
	libm.so.6 => /lib/libm.so.6 (0xb59f0000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xb59c0000)
	libffi.so.8 => /lib/libffi.so.8 (0xb59a0000)
```

Is there a way to fix this?

JeffyCN, 2022-04-12:
it that a 3rd party libmali?

try to normalize it using the scripts/normalize.sh

MastaG, 2022-04-12:
Ah yeah, that explains it :)
I didn't run normalize.sh yet, but I checked out the scripts.
It should also fix the ```.dynsym local symbol at index 3 (>= sh_info of 3)``` if I read it correctly.
It's now building lief using pip3 which seems to take a while.
I'll run the scripts and rebuild once it's done.

It's a not a 3rd party libmali but the user-space driver for ODroid XU3/XU4 from the ARM website for running on wayland/gbm.
It's pretty old and I've hex edited it to rename ```EGL_KHR_platform_gbm``` to ```EGL_MESA_platform_gbm``` for better compatibility.
But tossing it into your project makes up for all the missing gbm_ symbols :+1:

MastaG, 2022-04-12:
It seems to work now :+1: 
```
meson -Darch=arm -Dplatform=wayland -Dkhr-header=true -Doptimize-level=O3 -Dgpu=midgard-t62x -Dversion=r12p0 -Dsubversion=04rel0 ..
The Meson build system
Version: 0.60.3
Source dir: /mnt/sdcard/src/rockchip_mirrors
Build dir: /mnt/sdcard/src/rockchip_mirrors/build
Build type: native build
Project name: libmali
Project version: 1.9.0
C compiler for the host machine: ccache cc (gcc 12.0.1 "cc (GCC) 12.0.1 20220401 (Red Hat 12.0.1-0)")
C linker for the host machine: cc ld.bfd 2.37-24
Host machine cpu family: arm
Host machine cpu: armv7l
Message: Building for arm|midgard-t62x|r12p0|04rel0|wayland|O3
Message: Source libraries: ['optimize_3/arm-linux-gnueabihf/libmali-midgard-t62x-r12p0-04rel0-wayland.so']
Message: Provide wrappers
Library libmali-midgard-t62x-r12p0-04rel0-wayland found: YES
Run-time dependency threads found: YES
Library dl found: YES
Found pkg-config: /usr/bin/pkg-config (1.8.0)
Run-time dependency libdrm found: YES 2.4.110
Checking for function "gbm_create_device" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: YES 
Checking for function "gbm_bo_map" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_bo_unmap" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_bo_get_offset" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_bo_get_plane_count" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_device_get_format_modifier_plane_count" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_bo_get_handle_for_plane" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_bo_get_stride_for_plane" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_bo_get_fd_for_plane" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_bo_get_modifier" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_bo_create_with_modifiers" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_surface_create_with_modifiers" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_bo_get_bpp" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "gbm_format_get_name" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: NO 
Checking for function "eglCreateContext" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: YES 
Checking for function "gbm_create_device" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: YES (cached)
Checking for function "eglCreateContext" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: YES (cached)
Checking for function "eglCreateContext" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: YES (cached)
Checking for function "eglCreateContext" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: YES (cached)
Checking for function "wl_egl_window_create" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: YES 
Checking for function "clCreateContext" with dependency -llibmali-midgard-t62x-r12p0-04rel0-wayland: YES 
Program echo found: YES (/usr/bin/echo)
Program scripts/fixup_dummy.sh found: YES (/mnt/sdcard/src/rockchip_mirrors/scripts/fixup_dummy.sh)
Program scripts/fixup_nox11.sh found: YES (/mnt/sdcard/src/rockchip_mirrors/scripts/fixup_nox11.sh)
Build targets in project: 10

libmali 1.9.0

  User defined options
    arch          : arm
    gpu           : midgard-t62x
    khr-header    : true
    optimize-level: O3
    platform      : wayland
    subversion    : 04rel0
    version       : r12p0

Found ninja-1.10.2 at /usr/bin/ninja
Cleaning... 0 files.
```
After building:
```
$ ldd /usr/local/lib/libgbm.so.1
	linux-vdso.so.1 (0xbedda000)
	libmali_hook.so.1 => /usr/local/lib/libmali_hook.so.1 (0xb6ea0000)
	libc.so.6 => /lib/libc.so.6 (0xb6d20000)
	libmali.so.1 => /usr/local/lib/libmali.so.1 (0xb5c98000)
	libdrm.so.2 => /lib/libdrm.so.2 (0xb5c70000)
	/lib/ld-linux-armhf.so.3 (0xb6ee8000)
	librt.so.1 => /lib/librt.so.1 (0xb5c50000)
	libpthread.so.0 => /lib/libpthread.so.0 (0xb5c30000)
	libdl.so.2 => /lib/libdl.so.2 (0xb5c10000)
	libwayland-client.so.0 => /lib/libwayland-client.so.0 (0xb5bf0000)
	libwayland-server.so.0 => /lib/libwayland-server.so.0 (0xb5bd0000)
	libstdc++.so.6 => /lib/libstdc++.so.6 (0xb59f0000)
	libm.so.6 => /lib/libm.so.6 (0xb5990000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xb5960000)
	libffi.so.8 => /lib/libffi.so.8 (0xb5940000)
```

It now links with libmali.so.1 instead :+1:

closed 2022-04-12 by MastaG