JeffyCN/rockchip_mirrors #16: Need help downloading recent version of RK1808 SDK

kadirbeytorun, 2022-05-13:
Hey Jeffy,

We asked this on redmine also but I would like your opinion also since I am not sure if it will be solved properly on redmine.

We obtained RK1808 sdk from official rockchip repositories and successfully used them so far, but recently we noticed that some of the libraries (RGA, MPP, gstreamer-rockhip) are very old versions and have not been synced up to RK1808. 

We obtained RV1126 sdks also from rockchip repositories and observed that those same libraries are very recent versions and the sdk can be built for RV1126 config. However, it cannot be built for RK1808 config, so we cannot update from there also.

How could we get out of this dilemma? What would you suggest we do to find a solution that can work for us?

Cheers

JeffyCN, 2022-05-13:
we have a common develop sdk, and there are a few guys here to maintain different release branch for different chips.

the 1808 sdk should belong to hl@rock-chips.com, but he is busy with other things.

if you hit some compile error with common develop sdk, maybe i can help to check

kadirbeytorun, 2022-05-13:
Thanks for the speedy reply mate, you are the best.

We reached out to HL now, lets see what he says. Btw, how can we obtain the develop sdk? We cannot see that on github or rockchip servers?

kadirbeytorun, 2022-05-13:
I see, it seems we cannot reach original develop branch from rockchip servers then and we cannot use your mirror also since we need npu, camera drivers for kernel and filesystem.

So either FAE gives us an updated 1808 sdk or we have to stick with the old one.

Would it be possible to build  whole RK1808 buildroot config, from an updated sdk obtained for RK3399 or RV1126? I tried this with rv1126 yet there were errors when trying to build the whole thing, I can build some libraries though. We dont have access to 3399 official repositories, thus I couldn't try the same with that sdk.

JeffyCN, 2022-05-13:
it's possible to build from other chips' sdk, but there are a few repos of 1808 not in other sdk(for example npu), disabling those might work.

JeffyCN, 2022-05-13:
and new sdk might have bugs on 1808, since no one have tested on that chip for a long time.

anyway, i'm willing to debug some part of the sdk(maintained by me) an my free time, for example compile issues/weston/gstreamer(excluding mpp and kernel)/mali/scripts

kadirbeytorun, 2022-05-13:
You have been a great help mate, and I know you are doing it on your free time so I really appreciate it.

It is very sad that Rockchip has stopped developing on RK1808 and its sdk is outdated. I will wait for the reply from HL and also try to compile RV1126 sdk with RK1808 config. I will inform you of the results, so in case other people wonder about it you can use the feedback.

kadirbeytorun, 2022-05-18:
After reaching out to HL and through searching, it seems RK1808 sdk won't receive further updates and we decided not to update our sdk, but we can update different libraries whenever its necessary.

In my case, I had trouble with newest RGA library, I built it in RV1126 sdk with 1808 config, and then copied RGA kernel codes to old RK1808 sdk and built new kernel from it. As a result we get to use newest versions of RGA, MPP, gstreamer-rockchip libraries without updating our SDK. 

Thanks again for your help mate, cheers :)