JeffyCN/rockchip_mirrors #19: Gstreamer HEVC 10Bit issue on RK3588

avafinger, 2022-11-17:
Hi @JeffyCN ,
I cannot decode hevc-10bit.mkv on RK3588, while ffmpeg can decode just fine.

I have:
#define HAVE_NV12_10LE40 1

Do you need any additional info?

JeffyCN, 2022-11-17:
needs gst 1.16+(mostly used with gst 1.20)

avafinger, 2022-11-18:
Tested with gstreamer1.0-1.18.4 or do you mean rockchip1? rockchip1 is your 1.14.4, is there a branch with 1.16+?

JeffyCN, 2022-11-18:
gstreamer-rockchip requires gst at least 1.14.4, nv12 10b requires 1.16+.

and for h265, gst might needs extra patch, please check the buildroot or yocto sdk:
package/gstreamer1# find .|grep h265
./gst1-plugins-bad/0001-fix-h265_parser-read-vui-error.patch

root@RK3588:/# gst-play-1.0 /10b.ts --videosink=fakesink
Rga built version:156d31a9a8
Rga built version:156d31a9a8
Press 'k' to see a list of keyboard shortcuts.
Now playing /10b.ts
...
mpp[1558]: H265D_PARSER: extradata is encoded as hvcC format
mpp[1558]: mpp_buf_slot: set frame info: w 3840 h 2160 hor 3840 ver 2160
mpp[1558]: mpp_dec: setting default w 3840 h 2160 h_str 3840 v_str 2160
Redistribute latency...
0:00:05.4 / 0:00:04.7
Reached end of play list.

avafinger, 2022-11-18:
> and for h265, gst might needs extra patch,

h265 is working fine with 1.18.4. I will try to find and apply this patch.
Thanks

avafinger, 2022-11-18:
@JeffyCN 
I am not familiar with yocto / buildroot, care to share how i can pull the gstreamer plugins manually?
I found this reference: https://github.com/JeffyCN/yocto-conf/blob/master/include/multimedia.conf

avafinger, 2022-11-18:
@JeffyCN , your output from the gst-play command explicitly shows it is using Rga and mpp, in my pipeline i don't get this info, it that mean it is using sw decoding?

My pipeline:
sudol gst-launch-1.0 filesrc location=tos-4096x1720-tiles.mkv ! matroskademux name=demux demux.audio_0 ! queue ! decodebin ! audioconvert ! audioresample ! alsasink device="hw:0,0" demux.video_0 ! queue ! h265parse ! mppvideodec ! fakesink

JeffyCN, 2022-11-18:
check:
https://github.com/JeffyCN/rockchip_mirrors/tree/buildroot/package/gstreamer1

newer mpp might prefer to log to vsyslog than stdout

JeffyCN, 2022-11-18:
fyi, patches for 1.14/1.16/1.18 are maintained by Jiajian Wu <jair.wu@rock-chips.com>, if you are using those versions, might be better to acquire patches there

avafinger, 2022-11-18:
Ahh, the logs are in journal, but no reference to Rga, despite CONFIG_USE_RGA=1, the board is 16 GB RAM, i know that rga breaks if > 4GB, at least on ffmpeg. Maybe is not using rga for conversion, but i still get ~35% CPU usage (gstreamer) + ~40% CPU usage for xorg.

```
Nov 17 15:30:09 rock5b mpp[2108]: mpp_info: mpp version: ec84649 author: Yandong Lin    2022-11-17 [h265d_vdpu34x]: fix dec output green img with err source
Nov 17 15:30:09 rock5b mpp[2108]: mpp_info: mpp version: ec84649 author: Yandong Lin    2022-11-17 [h265d_vdpu34x]: fix dec output green img with err source
Nov 17 15:30:09 rock5b mpp[2108]: mpp_info: mpp version: ec84649 author: Yandong Lin    2022-11-17 [h265d_vdpu34x]: fix dec output green img with err source
Nov 17 15:30:09 rock5b mpp[2108]: hal_vp8e_api_v2: hal_vp8e_init Failed to init due to unsupported hard mode, hw_flag = 269034010
Nov 17 15:30:09 rock5b mpp[2108]: mpp_enc_hal: mpp_enc_hal_init hal hal_vp8e init failed ret -1002
Nov 17 15:30:09 rock5b mpp[2108]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 9
Nov 17 15:30:09 rock5b mpp[2108]: mpp_enc: mpp_enc_init_v2 could not init enc hal
Nov 17 15:30:09 rock5b mpp[2108]: mpp: error found on mpp initialization
Nov 17 15:30:09 rock5b mpp[2108]: mpp_info: mpp version: ec84649 author: Yandong Lin    2022-11-17 [h265d_vdpu34x]: fix dec output green img with err source
Nov 17 15:30:09 rock5b mpp[2108]: mpp_info: mpp version: ec84649 author: Yandong Lin    2022-11-17 [h265d_vdpu34x]: fix dec output green img with err source
Nov 17 15:30:09 rock5b mpp[2108]: mpp_buf_slot: set frame info: w 1920 h 1080 hor 2304 ver 1080
Nov 17 15:30:09 rock5b mpp[2108]: mpp_dec: setting default w 1920 h 1080 h_str 2304 v_str 1080
Nov 17 15:30:09 rock5b mpp[2108]: H265D_PARSER: extradata is encoded as hvcC format

```

JeffyCN, 2022-11-18:
for x11 + gst, the prefered way would be rkximagesink(zero copy drm display) or hacked xvimagesink(zero copy+1 or 2 gpu render)

avafinger, 2022-11-18:
I get a black screen with the current setup using rkximagesink, this is without that patch, but for any h264 also.
Would you know why vp8 enc hal is not initialized?

Nov 17 15:30:09 rock5b mpp[2108]: mpp_enc_hal: mpp_enc_hal_init hal hal_vp8e init failed ret -1002
Nov 17 15:30:09 rock5b mpp[2108]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 9

JeffyCN, 2022-11-19:
black screen might due to kernel or wrong plane
hal issue might due to kernel.

3588's newest kernel needs to buy sdk license to acquire(i've heard that), and the display & mpp drivers changed a lot, it's possible to have issues in old version

closed 2022-11-25 by JeffyCN