JeffyCN/rockchip_mirrors #20: Fetch error during compilation

TYeniyayla, 2022-11-29:
I'm only getting an error when pulling from this address. Any idea about the reason for this?

```
Initialising tasks: 100% |#######################################################| Time: 0:00:09
Sstate summary: Wanted 4778 Local 0 Mirrors 0 Missed 4778 Current 681 (0% match, 12% complete)
NOTE: Executing Tasks
WARNING: rockchip-libmali-1.0-r0 do_fetch: Failed to fetch URL git://github.com/JeffyCN/mirrors.git;protocol=https;branch=libmali;, attempting MIRRORS if available
ERROR: rockchip-libmali-1.0-r0 do_fetch: Fetcher failure: Unable to find revision 71900e3e0c47f6191b44be9ecd37ffc194af6687 in branch libmali even from upstream
ERROR: rockchip-libmali-1.0-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/JeffyCN/mirrors.git;protocol=https;branch=libmali;')
ERROR: rockchip-librga-1.0-r0 do_unpack: Bitbake Fetcher Error: UnpackError('No up to date source found: clone directory not available or not up to date: /media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject-kirkstone/build/downloads/git2/github.com.JeffyCN.mirrors.git; shallow clone not enabled', 'git://github.com/JeffyCN/mirrors.git;protocol=https;branch=linux-rga;')
ERROR: Logfile of failure stored in: /media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject-kirkstone/build/tmp/work/mepsan_pcr172-mepos-linux-gnueabi/rockchip-libmali/1.0-r0/temp/log.do_fetch.22862
ERROR: Logfile of failure stored in: /media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject-kirkstone/build/tmp/work/armv7vet2hf-neon-mepos-linux-gnueabi/rockchip-librga/1.0-r0/temp/log.do_unpack.22884
ERROR: Task (/media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject-kirkstone/meta-rockchip/recipes-graphics/rockchip-libmali/rockchip-libmali.bb:do_fetch) failed with exit code '1'
ERROR: Task (/media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject-kirkstone/meta-rockchip/recipes-graphics/rockchip-librga/rockchip-librga.bb:do_unpack) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2585 tasks of which 2583 didn't need to be rerun and 2 failed.

Summary: 2 tasks failed:
  /media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject-kirkstone/meta-rockchip/recipes-graphics/rockchip-libmali/rockchip-libmali.bb:do_fetch
  /media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject-kirkstone/meta-rockchip/recipes-graphics/rockchip-librga/rockchip-librga.bb:do_unpack
Summary: There was 1 WARNING message.
Summary: There were 3 ERROR messages, returning a non-zero exit code.
```

JeffyCN, 2022-11-29:
oops, that was a copy-paste mistake, fixing it right now

TYeniyayla, 2022-11-29:
It's working properly now. Thank you. passed that error.

closed 2022-12-01 by JeffyCN