JeffyCN/rockchip_mirrors #21: [RK3588] Single gstreamer pipeline not able to hit 4K@120 fps, but two separate pipelines hit 4K@60 fps

lgarbarini, 2023-01-20:
I have a gstreamer pipeline either pulling for USB/MIPI/file I expect to be able to achieve 4K@120 FPS. However, I seem to be unable to get a single encoder to achieve 4K@120 fps. However, two pipelines at the same time are able to achieve >120 fps combined.  

I have tried with both live and test sources and the results seem to be the same.

8k@30 fps, works as expected (~29 fps average):
```shell
gst-launch-1.0 videotestsrc ! video/x-raw,framerate=6/1 ! videorate ! queue ! video/x-raw,width=7680,height=4320,framerate=60/1,format=NV12 ! mpph265enc ! fpsdisplaysink video-sink=fakesink text-overlay=false -ve
```

1080P@240 fps, works as expected (~230 fps average):
```shell
gst-launch-1.0 videotestsrc ! video/x-raw,framerate=30/1 ! videorate ! queue ! video/x-raw,width=1920,height=1080,framerate=240/1,format=NV12 ! mpph265enc ! fpsdisplaysink video-sink=fakesink text-overlay=false -ve
```

4K@120 fps, Doesn't work as expected (63 fps average):
```shell
gst-launch-1.0 videotestsrc ! video/x-raw,framerate=30/1 ! videorate ! queue ! video/x-raw,width=3840,height=2160,framerate=120/1,format=NV12 ! mpph265enc ! fpsdisplaysink video-sink=fakesink text-overlay=false -ve
```

However, if I run 2x of the above 4K pipelines simultaneously, both achieve 4K@60 fps (120 fps total):
```shell
# In terminal 1:
gst-launch-1.0 videotestsrc ! video/x-raw,framerate=30/1 ! videorate ! queue ! video/x-raw,width=3840,height=2160,framerate=120/1,format=NV12 ! mpph265enc ! fpsdisplaysink video-sink=fakesink text-overlay=false -ve
# results in ~63 fps average

# In terminal 2:
gst-launch-1.0 videotestsrc ! video/x-raw,framerate=30/1 ! videorate ! queue ! video/x-raw,width=3840,height=2160,framerate=120/1,format=NV12 ! mpph265enc ! fpsdisplaysink video-sink=fakesink text-overlay=false -ve
# results in ~63 fps average
```

[image]

lgarbarini, 2023-01-20:
Looks like setting `auto_tile` to `1` enables the use of both cores.

https://github.com/JeffyCN/rockchip_mirrors/blob/546cb913e3feabc4a41a6220fa80d81e0dea71c2/gst/rockchipmpp/gstmpph265enc.c#L185

```
diff --git a/gst/rockchipmpp/gstmpph265enc.c b/gst/rockchipmpp/gstmpph265enc.c
index 2605f0f..4ce315e 100644
--- a/gst/rockchipmpp/gstmpph265enc.c
+++ b/gst/rockchipmpp/gstmpph265enc.c
@@ -182,6 +182,7 @@ gst_mpp_h265_enc_apply_properties (GstVideoEncoder * encoder)
   if (G_LIKELY (!mppenc->prop_dirty))
     return TRUE;

+  mpp_enc_cfg_set_s32 (mppenc->mpp_cfg, "h265:auto_tile", 1);
   mpp_enc_cfg_set_s32 (mppenc->mpp_cfg, "h265:qp_init", self->qp_init);

   if (mppenc->rc_mode == MPP_ENC_RC_MODE_FIXQP) {
```

JeffyCN, 2023-12-01:
not sure about the "auto_tile" @HermanChen 
but i've uploaded a patch in gst-mpp for async encoding.