JeffyCN/rockchip_mirrors #25: Failed to initialize MPP context?

danergo, 2023-04-17:
I compiled these repos: librga (from rockchip_mirrors), mpp (this repo), ffmpeg-rk on a NanoPI R5C locally.
Then I also did "sudo make install" for all projects.

Now, when I try running this ffmpeg I got:

```
[hevc_rkmpp @ 0x55a919c7e0] Failed to initialize MPP context (code = -1).
[hevc_rkmpp @ 0x55a919c7e0] Failed to initialize RKMPP decoder.                                                       

```
Can you help me understand which point is missing?

JeffyCN, 2023-04-17:
i don't use ffmpeg much, i would prefer gstreamer.

for ffmpeg, maybe you can check:
https://github.com/JeffyCN/FFmpeg/issues/18

danergo, 2023-04-17:
I wish to use Emby (it's a media server), which relies on FFMPEG, so unfortunately I don't have other choice.

Will try yours, and let you know. Thank you very much!

danergo, 2023-04-17:
My mpp might be wrong, can you please point me to the correct mpp repo which I need to build?

JeffyCN, 2023-04-18:
the mpp should match the kernel driver, so i don't know which version you need...

the mpp maintainer's repo:
https://github.com/HermanChen/mpp

amazingfate, 2023-04-18:
You can check mpp logs from syslog by `journalctl -f`.
I encountered this issue when runnning as root on armbian because armbian has a kernel which has enabled `CONFIG_ION=y` and then mpp will try to use `/dev/ion` because the process just have the permission to access it. Running as normal user can avoid it.

closed 2023-04-19 by JeffyCN

jjm2473, 2023-04-22:
On rk3568 , `/dev/ion` can break MPP, just delete it, or disable it on kernel config.

DMA_HEAP has better performance than ION, it's a perfer solution. If DMA_HEAP not supported on your kernel (like 4.19), use DRM instead.

danergo, 2023-04-22:
If I remove /Dev/ion, it works. Thank you.