JeffyCN/rockchip_mirrors #26: Gstreamer videoconvert / videoscale support

avafinger, 2023-04-25:
Hi @JeffyCN ,

I would like to scale a video but haven't found a way to do so. Previous versions had rgaconvert where you could scale the image.
Interesting that if i scale the image to same size it works, For example, a 1920x1080 video:

`gst-launch-1.0 -v filesrc location=~/Videos/jellyfish-5-mbps-hd-h264.mkv ! matroskademux ! h264parse ! mppvideodec ! videoconvert ! video/x-raw,format=NV12,width=1920,height=1080 ! glimagesink
`

Tried with videoscale:

`gst-launch-1.0 -v filesrc location=~/Videos/jellyfish-5-mbps-hd-h264.mkv ! matroskademux ! h264parse ! mppvideodec ! videoscale ! video/x-raw,width=1920,height=1080 ! glimagesink`

If a scale like so:

`gst-launch-1.0 -v filesrc location=~/Videos/jellyfish-5-mbps-hd-h264.mkv ! matroskademux ! h264parse ! mppvideodec ! videoscale ! video/x-raw,width=1920,height=540 ! glimagesink
`

I get an error.

Can you please provide the correct way to scale the image, or there is no support?

JeffyCN, 2023-04-26:
i've added rga scale support in mppvideodec (width/hight properties), but the rga might not working on some chips (due to API changes)

you can try it, or maybe try to the convert code to new rga API in mppvideodec

avafinger, 2023-04-26:
I missed that, thank you!

gst-launch-1.0 -v filesrc location=~/Videos/jellyfish-5-mbps-hd-h264.mkv ! matroskademux ! h264parse ! mppvideodec width=1920 height=720 !  glimagesink

JeffyCN, 2023-04-26:
try kmssink

or maybe xvimagesink or waylandsink

avafinger, 2023-04-26:
They all failed. I need to display several decoded videos, in a chessboard way. it is only possible with glimagesink, maybe waylandsink could do better but currently, i am on X11. Do you think waylandsink can draw the video on a screen position?

JeffyCN, 2023-04-26:
xvimagesink should work

waylandsink should be ok in rockchip's buildroot sdk, needs lots of hacks

for glimagesink, maybe you can ask the maintainer?

avafinger, 2023-04-26:
Sorry for the repetitive question, I tried xvimagesink as below but failed (maybe the properties to pass are not correct, if you have an idea what is wrong).

**xvimagesink**

```
rock@rock5b:~$ gst-launch-1.0 -v filesrc location=~/Videos/jellyfish-5-mbps-hd-h264.mkv ! matroskademux ! h264parse ! mppvideodec width=320 height=240 ! rkximagesink sink_1::xpos=0 sink_1::ypos=0 
WARNING: erroneous pipeline: no property "sink_1::xpos" in element "rkximagesink0"
rock@rock5b:~$ gst-launch-1.0 -v filesrc location=~/Videos/jellyfish-5-mbps-hd-h264.mkv ! matroskademux ! h264parse ! mppvideodec width=320 height=240 ! rkximagesink render-rectangle='<120,0,320,240>'
WARNING: erroneous pipeline: no property "render-rectangle" in element "rkximagesink0"
rock@rock5b:~$ gst-launch-1.0 -v filesrc location=~/Videos/jellyfish-5-mbps-hd-h264.mkv ! matroskademux ! h264parse ! mppvideodec width=320 height=240 ! rkximagesink sink_0::xpos=0 sink_0::ypos=0 
WARNING: erroneous pipeline: no property "sink_0::xpos" in element "rkximagesink0"

```

**kmssink:**

```
rock@rock5b:~$ gst-launch-1.0 -v filesrc location=~/Videos/jellyfish-5-mbps-hd-h264.mkv ! matroskademux ! h264parse ! mppvideodec width=320 height=240 ! kmssink sink_1::xpos=0 sink_1::ypos=0 
WARNING: erroneous pipeline: no property "sink_1::xpos" in element "kmssink0"

```

**glimagesink ignored the properties:**

```
rock@rock5b:~$ gst-launch-1.0 -v filesrc location=~/Videos/jellyfish-5-mbps-hd-h264.mkv ! matroskademux ! h264parse ! mppvideodec width=320 height=240 ! glimagesink sink_1::xpos=0 sink_1::ypos=0 
Setting pipeline to PAUSED ...
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '5'.
Pipeline is PREROLLING ...
Got context from element 'sink': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayX11\)\ gldisplayx11-0";
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, level=(string)4, profile=(string)high, codec_data=(buffer)01640028ffe1002e67640028ac2ca401e0089f97ff0001000152020202800001f4800075307110001312c0000e4e1df8c7076858b44801000568eb735250, stream-format=(string)avc, alignment=(string)au, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30000/1001, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)bt709, parsed=(boolean)true
/GstPipeline:pipeline0/GstMppVideoDec:mppvideodec0.GstPad:sink: caps = video/x-h264, level=(string)4, profile=(string)high, codec_data=(buffer)01640028ffe1002e67640028ac2ca401e0089f97ff0001000152020202800001f4800075307110001312c0000e4e1df8c7076858b44801000568eb735250, stream-format=(string)avc, alignment=(string)au, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30000/1001, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)bt709, parsed=(boolean)true
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = video/x-h264, level=(string)4, profile=(string)high, codec_data=(buffer)01640028ffe1002e67640028ac2ca401e0089f97ff0001000152020202800001f4800075307110001312c0000e4e1df8c7076858b44801000568eb735250fdf8f800, stream-format=(string)avc, alignment=(string)au, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30000/1001
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, level=(string)4, profile=(string)high, codec_data=(buffer)01640028ffe1002e67640028ac2ca401e0089f97ff0001000152020202800001f4800075307110001312c0000e4e1df8c7076858b44801000568eb735250, stream-format=(string)avc, alignment=(string)au, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30000/1001, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)bt709, parsed=(boolean)true
/GstPipeline:pipeline0/GstMppVideoDec:mppvideodec0.GstPad:sink: caps = video/x-h264, level=(string)4, profile=(string)high, codec_data=(buffer)01640028ffe1002e67640028ac2ca401e0089f97ff0001000152020202800001f4800075307110001312c0000e4e1df8c7076858b44801000568eb735250, stream-format=(string)avc, alignment=(string)au, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30000/1001, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)bt709, parsed=(boolean)true
/GstPipeline:pipeline0/GstMppVideoDec:mppvideodec0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)NV12, width=(int)320, height=(int)240, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)30000/1001
/GstPipeline:pipeline0/GstGLImageSinkBin:glimagesinkbin0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw(memory:DMABuf), format=(string)NV12, width=(int)320, height=(int)240, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)30000/1001
[... cut ...] iew-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)30000/1001
Pipeline is PREROLLED ...0 %)
Prerolled, waiting for async message to finish...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
^Chandling interrupt. (14.1 %)
Interrupt: Stopping pipeline ...
Execution ended after 0:00:04.253642222
Setting pipeline to NULL ...
Freeing pipeline ...

```

JeffyCN, 2023-04-27:
i would prefer xvimagesink, because rkximagesink is kind like kmssink, it requires hw drm plane, and not work well with zpos.

for slow xvimagesink, maybe you can try the rockchip custom xserver and xvimagesink

please check my xserver repo and meta-rockchip for patches

JeffyCN, 2023-04-27:
the kmssink and xvimagesink should support render-rectangle, and i've checked the glimagesink source code in 1.22, it should support that too on x11 platform

avafinger, 2023-04-27:
> you can try the rockchip custom xserver and xvimagesink

Thank you for your reply.

I am using gstreamer 1.20.3 (patched) and your custom xserver and no way to position the image.
The problem is xvimagesink consumes 87% CPU (Xorg) while rkximage 35% CPU (Xorg).
glimagesink is able to render it fine, but positioning the rendered image is ignored, and yet i have to find a way NOT to draw the window border and title.

I will try with 1.22 or get back to ffmpeg.

JeffyCN, 2023-04-27:
the custom xvimagesink should be zero-copied, and it has a draw border prop.

glimagesink should support render-rectangle='<x,y,w,h>'

JeffyCN, 2023-04-27:
i haven't got a device to test it, but the glimagesink x11's source code doesn't look like having border

avafinger, 2023-04-27:
> he custom xvimagesink should be zero-copied

Is this Hack here?
https://github.com/JeffyCN/meta-rockchip/blob/master/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.22/0007-HACK-xvimagesink-Support-dma-buffer-rendering.patch

JeffyCN, 2023-04-27:
the command seems fine to me.

that should work with hacked xserver:
https://github.com/JeffyCN/xorg-xserver/blob/1.20.4/glamor/glamor_xv.c#L464

JeffyCN, 2023-04-27:
it works well on rockchip official sdk.

you can add log to make sure the glamor_xv_render_dma success

JeffyCN, 2023-04-28:
i've got a debian 11 device, assuming you are using openbox too.

about position:
there're multiple ways to set position:
1/ use openbox's per-app config(you can set decoration here too):
https://unix.stackexchange.com/questions/656277/openbox-open-new-windows-in-specific-locations
2/ add size hint to gst sink (openbox honors that):
https://github.com/wilsonlab/mwsoft64/blob/60bfd5cb787da756c45e566322baff1926e1870c/src/xview/init.c#L291
3/ use XMoveResizeWindow and disable openbox
xvimagesink supports render-rectangle, but it's a overlay render-rectangle (doesn't change window size or position), you can use XMoveResizeWindow in it when xwindow_id < 0(toplevel window)
glimagesink supports render-rectangle, and it's using XMoveResizeWindow, but openbox move it again after that.

and about decoration:
1/ use openbox's per-app config(you can set decoration here too):
https://unix.stackexchange.com/questions/656277/openbox-open-new-windows-in-specific-locations
2/ add motif hint to gst sink (openbox honors that):
xvimagesink already has it, you can set it to 0 to disable decoration:
```
+++ b/sys/xvimage/xvcontext.c
@@ -1098,7 +1098,7 @@ gst_xvcontext_create_xwindow (GstXvContext * context, gint width, gint height)
     hints = g_malloc0 (sizeof (MotifWmHints));

     hints->flags |= MWM_HINTS_DECORATIONS;
-    hints->decorations = 1 << 0;
+    hints->decorations = 0;//1 << 0;
```
you can port it to glimagesink too.

closed 2023-04-28 by JeffyCN

JeffyCN, 2023-04-28:
https://github.com/JeffyCN/rockchip_mirrors/commit/8cf3403a9e07ed703273ab54377fb2c007da14e0

avafinger, 2023-04-29:
@JeffyCN 

I applied your patch manually (on 1.20.1) and tested the following situation:

Fired 1:
gst-launch-1.0 videotestsrc ! xvimagesink render-rectangle='<0,200,320,240>' decorations=0

Fired 2:
gst-launch-1.0 filesrc location=~/Videos/jellyfish-5-mbps-hd-h264.mkv ! matroskademux ! h264parse ! mppvideodec width=320 height=240 ! xvimagesink render-rectangle='<0,0,320,240>' decorations=0

decorations=0 is working despite Xorg still drawing shadows around the frame (glamor), but that's ok.

What i wanted to achieve:

1. Render video  on (0,0) (320,240) (x,y) (dx,dy)
2. Render videotest (0,200) (320,240)  (x,y) (dx,dy)

```
0,0
+--------------+
|    video     |
|              |
+--------------+ (dx,dy) 320,240
|  testsrc     |
|              |
+--------------+

```
render-rectangle='<0,200,320,240> clipped the client (320,40) as seen in the image.
[image]

Another issue is that gst_video_center_rect is still being called and assertion is raised
**src->h != 0**

Maybe I expressed myself wrong, i wanted this:

```
+--------------+--------------+--------------+--------------+--------------+--------------+--------------+
| video 1      | video 2      |video 3       |video 4       |  video 5     |video 6       |  video 7     |   
+--------------+--------------+--------------+--------------+--------------+--------------+--------------+

```

Can you see if you get the same behavior, so i know i need to review the patch?

JeffyCN, 2023-04-29:
i'll be back to work in 5 days later

and maybe we can create a temporary repo for xvimagesink
and maybe you can try Qt(a big window with a few small videos inside)

JeffyCN, 2023-04-29:
https://github.com/JeffyCN/rockchip_mirrors/commit/ca1426180c3b383139c298432ab9d657d13ac844

avafinger, 2023-04-29:
works!
[image]

Just a small issue, if you hit **CTRL+C** (someone call **gst_video_center_rect**):

**^C**handling interrupt. (8.1 %)
Interrupt: Stopping pipeline ...
Execution ended after 0:00:02.421331022
Setting pipeline to NULL ...

```
(gst-launch-1.0:93415): GStreamer-Video-CRITICAL **: 16:15:58.563: gst_video_center_rect: assertion 'src->h != 0' failed
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  151 (XVideo)
  Minor opcode of failed request:  19 ()
  Value in failed request:  0x61
  Serial number of failed request:  1840
  Current serial number in output stream:  1841

```

JeffyCN, 2023-04-30:
the center help would be called on every frame, but i have no idea why the src size been empty, could related to gst mppdec or mpp.

as a workaround, maybe you can just return earlier when src is empty in xvimagesink's put_image flow.

i'll check it when back to work too

avafinger, 2023-04-30:
Sometimes it happens on EOS like this:

Got EOS from element "pipeline0".
Execution ended after 0:02:30.442526953
Setting pipeline to NULL ...

(gst-launch-1.0:18541): GStreamer-Video-CRITICAL **: 22:27:32.566: gst_video_center_rect: assertion 'src->h != 0' failed
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  151 (XVideo)
  Minor opcode of failed request:  19 ()
  Value in failed request:  0x6e
  Serial number of failed request:  81035
  Current serial number in output stream:  81036

JeffyCN, 2023-05-04:
```
+++ b/sys/xvimage/xvimagesink.c
@@ -1690,6 +1690,11 @@ gst_xv_image_sink_expose (GstVideoOverlay * overlay)

   GST_DEBUG ("doing expose");
   gst_xv_image_sink_xwindow_update_geometry (xvimagesink);
+
+  if (GST_VIDEO_SINK_WIDTH (xvimagesink) <= 0 ||
+      GST_VIDEO_SINK_HEIGHT (xvimagesink) <= 0)
+    return;
+
   gst_xv_image_sink_xvimage_put (xvimagesink, NULL);
 }
```

there's a ConfigureNotify on EOS that might trigger a expose-redraw

avafinger, 2023-05-05:
Thanks, that fixed the issue.

avafinger, 2023-05-08:
Hi @JeffyCN ,

i still cannot get enough performance with xvimagesink compared to glimagesink in terms of CPU usage for Xorg it should be 0-copy but i get it only on glimagesink.
NVidea has glimagesink window_x=VAL window_y=VAL , i tried to create two properties like:
glimagesink xpos=100 ypos=200, following their code, but i ended with the client draw position, not the window.
I tried to follow your implementation for xvwindow but gstglimagesink.c and gstglwindow_x11.c is quite different and i don't have experience with gstreamer.

Any chance you revisit glimagesink and add xpos,ypos and decoration properties? render-rectangle is ignored for the same reason above, so ypos and xpos would be enough.

avafinger, 2023-05-08:
@JeffyCN 
I missed your glimagesink patch, please ignore my request, i will test it. Thanks

avafinger, 2023-05-08:
@JeffyCN 

I applied the https://github.com/JeffyCN/rockchip_mirrors/blob/7fe4cc10ae131a4c212740cc836d372317427395/package/gstreamer1/gst1-plugins-base/0020-gl-x11-Allow-disabling-decorations-for-toplevel-wind.patch

but i don't find where you add the decorations properties in glimagesink.

**Here is my pipeline:**
DISPLAY=:0.0 gst-launch-1.0 filesrc location=~/Videos/heavenknowswhat-tlr1_h1080p.mov ! qtdemux ! h264parse ! mppvideodec ! glimagesink render-rectangle='<1440,810,480,270>' decorations=0
WARNING: erroneous pipeline: no property "decorations" in element "glimagesinkbin0"

JeffyCN, 2023-05-09:
you can check the patch, there's an environment setting added

yrik, 2023-09-01:
Regarding hardware support for videoconvert and videoscale, I have a case where I get raw video on the input and then send it further as a raw video, that's why I can not use width/height props of encode/decode flow.