JeffyCN/rockchip_mirrors #28: gstreamer NV24 support

avafinger, 2023-06-01:
Hi @JeffyCN ,

Is there support for NV24  and which version? Can you share a pipeline?

JeffyCN, 2023-06-01:
there's a commit about NV24:

commit 98da85ade042d8495a89f67ea164529f7c5918
55                                           Author: Nicolas Dufresne <nicolas.dufresne@co
llabora.com>
Date:   Fri Sep 17 15:39:54 2021 -0400           kmssink: Add NV24 support

    This was tested on RK3566 platform, using vendor DRM driver.
    Part-of: <https://gitlab.freedesktop.org/
gstreamer/gst-plugins-bad/-/merge_requests/25
30>

avafinger, 2023-06-01:
I applied that patch already (gst-plugins-bad1-1.20.3), but it failed to convert NV24 to NV12 using RGA. RK3588. That's why i asked you for a sample pipeline.

I wanted to capture video from HDMI (rk3588, board A) to HDMI-RX-IN (rk3588, board B) but for some reason, the hand-shake chose NV24 instead of BGR3, and i can't change it. 
In Kernel 5.10.66 i have the correct pixel format (BGR3), but not in 5.10.110 (NV24). It was even possible to make a loop between the HDMI out and HDMI In on kernel 5.10.66 to record video.

I cheated the HDMI-RX-IN with a capture device so i get BGR3 so i can capture a mkv video.
that's why i don't have the full error saved, but it asked me to fill a bug report to gstreamer.

**off-topic:**
Another issue i have is if the board have HDMI-1 and HDMI-2 while using Weston (wayland) the gstreamer output video to HDMI-A-1 but gstreamer capture video from HDMI-A-2. 

Anyway, is there a way to force HDMI out to choose NV12 when connected directly to HDMI-RX-IN? Any possible trick you would know?

avafinger, 2023-06-02:
If you have a working pipeline for NV24 on rk3588, please suggest it.
Also, if there is a way to choose the device to output the video stream in gstreamer, please suggest it.

I will close it now.

closed 2023-06-02 by avafinger