JeffyCN/rockchip_mirrors #31: mppjpegdec format=RGB gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed

NIKovachev, 2023-07-24:
Hi,
mppjpegdec format=RGB errors and corrupts output frames when using 4K frames.  Other formats works fine (RGBA for example).

Error example:
```
root@firefly:~/sensevision# gst-launch-1.0 -v v4l2src device=/dev/video12 ! image/jpeg,width=3840,height=2160,framerate=30/1 ! mppjpegdec format=RGB ! fakesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)RGB, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:1:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)RGB, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:1:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1

(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.052: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
0:00:00.1 / 99:99:99.
(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.098: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed

(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.153: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
0:00:00.4 / 99:99:99.
(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.194: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed

(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.254: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
0:00:00.6 / 99:99:99.
(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.351: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
0:00:00.6 / 99:99:99.
(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.375: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed

(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.434: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
0:00:00.7 / 99:99:99.
(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.496: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
0:00:00.8 / 99:99:99.
(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.593: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed

(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.619: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
0:00:00.9 / 99:99:99.
(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.674: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
0:00:01.0 / 99:99:99.
(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.775: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed

(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.796: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed

(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.854: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
0:00:01.2 / 99:99:99.
(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:24.916: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
0:00:01.2 / 99:99:99.
(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:25.019: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed

(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:25.039: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
0:00:01.3 / 99:99:99.
(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:25.094: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:01.612582624
Setting pipeline to NULL ...

(gst-launch-1.0:396673): GStreamer-CRITICAL **: 19:18:25.194: gst_buffer_resize_range: assertion 'bufmax >= bufoffs + offset + size' failed
Freeing pipeline ...
root@firefly:~/sensevision# 

```

Working example:
```
root@firefly:~/sensevision# gst-launch-1.0 -v v4l2src device=/dev/video12 ! image/jpeg,width=3840,height=2160,framerate=30/1 ! mppjpegdec format=RGBA ! fakesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)RGBA, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:1:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)RGBA, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:1:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:02.544840143
Setting pipeline to NULL ...
Freeing pipeline ...
root@firefly:~/sensevision# 
```

JeffyCN, 2023-07-25:
oops, please  try this:
https://github.com/JeffyCN/mirrors/commit/e0670aa3054458a737dde18f470a0d50de6b924d

JeffyCN, 2023-08-03:
guessing it's fixed.

closed 2023-08-03 by JeffyCN

NIKovachev, 2023-12-05:
I finally test is and it seems to have some issues:

root@firefly:~/mirrors-gstreamer-rockchip/builddir# gst-launch-1.0 -v --gst-debug=2 shmsrc socket-path=/var/ramfs/camera_shm ! image/jpeg,width=3840,height=2160,framerate=20/1 ! mppjpegdec format=BGR dma-feature=0 max-errors=-1 ! fpsdisplaysink video-sink=fakesink text-overlay=false -ve
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)20/1, parsed=(boolean)true
0:00:00.066022694 19115   0x55b5dccc60 WARN              video-info video-info.c:196:validate_colorimetry: color matrix RGB is only supported with RGB format, NV12 is not
0:00:00.066087444 19115   0x55b5dccc60 WARN              video-info video-info.c:520:gst_video_info_from_caps: invalid colorimetry, using default
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:src: caps = video/x-raw, format=(string)BGR, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw, format=(string)BGR, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-raw, format=(string)BGR, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink: caps = video/x-raw, format=(string)BGR, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
0:00:00.066738737 19115   0x55b5dccc60 WARN            videodecoder gstvideodecoder.c:2573:gst_video_decoder_chain:<mppjpegdec0> Received buffer without a new-segment. Assuming timestamps start from 0.
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)20/1, parsed=(boolean)true
0:00:00.079270216 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:591:gst_mpp_dec_get_frame:<mppjpegdec0> MPP is not able to generate pts
0:00:00.079528633 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.079565092 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.134382017 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.134503934 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.195123160 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.195225536 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.255321720 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.255417970 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.315716904 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.315802071 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.375451921 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.375527463 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.435773730 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.435864147 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.496849833 19115   0x55b5dcd180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
[... cut ...]
0:00:01.210566862 19115   0x55b5dccc60 WARN            videodecoder gstvideodecoder.c:1246:gst_video_decoder_sink_event_default:<mppjpegdec0> error: no valid frames found
ERROR: from element /GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0: No valid frames decoded before end of stream
Additional debug info:
../gst-libs/gst/video/gstvideodecoder.c(1246): gst_video_decoder_sink_event_default (): /GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0:
no valid frames found
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...
root@firefly:~/mirrors-gstreamer-rockchip/builddir#