JeffyCN/rockchip_mirrors #32: mppjpegdec hangs after several minutes on RK3588

NIKovachev, 2023-08-01:
Hi,
after running decode with mppjpegdec for several minutes it errors and on restart the below appears. I'm using RK3588 ( Firefly AIO-3588Q)

**Initial error:**
```
0:01:44.886656827  5950   0x55b7a95460 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<v4l2src0> error: Internal data stream error.
0:01:44.886713411  5950   0x55b7a95460 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<v4l2src0> error: streaming stopped, reason error (-5)
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason error (-5)
Execution ended after 0:01:44.796785884
Setting pipeline to NULL ...
0:01:45.118470525  5950   0x55b7a954c0 WARN                 shmsink gstshmsink.c:840:pollthread_func:<shmsink0> error: Failed waiting on fd activity
0:01:45.118545776  5950   0x55b7a954c0 WARN                 shmsink gstshmsink.c:840:pollthread_func:<shmsink0> error: gst_poll_wait returned -1, errno: 16
ERROR: from element /GstPipeline:pipeline0/GstShmSink:shmsink0: Failed waiting on fd activity
Additional debug info:
../sys/shm/gstshmsink.c(840): pollthread_func (): /GstPipeline:pipeline0/GstShmSink:shmsink0:
gst_poll_wait returned -1, errno: 16
Freeing pipeline ...
root@firefly:~# 
```

**When I try to restart:**
```
root@firefly:~# gst-launch-1.0 -v --gst-debug=2 v4l2src device=/dev/video12 ! image/jpeg,width=3840,height=2160,framerate=30/1 ! mppjpegdec ! queue ! fakesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)NV12, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)2:4:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstQueue:queue0.GstPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)NV12, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)2:4:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstQueue:queue0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)NV12, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)2:4:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)NV12, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)2:4:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
0:00:00.362207292  6497   0x55bee02400 WARN          v4l2bufferpool gstv4l2bufferpool.c:814:gst_v4l2_buffer_pool_start:<v4l2src0:pool0:src> Uncertain or not enough buffers, enabling copy threshold
0:00:00.741355943  6497   0x55bee02400 ERROR                 mppdec gstmppdec.c:1012:gst_mpp_dec_handle_frame:<mppjpegdec0> failed to send packet
0:00:00.741474361  6497   0x55bee02400 WARN                  mppdec gstmppdec.c:1016:gst_mpp_dec_handle_frame:<mppjpegdec0> can't handle this frame
0:00:00.743893758  6497   0x55bee02400 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<v4l2src0> error: Internal data stream error.
0:00:00.743975426  6497   0x55bee02400 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<v4l2src0> error: streaming stopped, reason error (-5)
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason error (-5)
Execution ended after 0:00:00.506613020
Setting pipeline to NULL ...
0:00:00.761352422  6497   0x55bee05860 WARN                  mppdec gstmppdec.c:563:gst_mpp_dec_get_frame:<mppjpegdec0> MPP is not able to generate pts
Freeing pipeline
```

**Running the same pipe but with jpegdec (works fine)**
```
root@firefly:~# gst-launch-1.0 -v --gst-debug=2 v4l2src device=/dev/video12 ! image/jpeg,width=3840,height=2160,framerate=30/1 ! jpegdec ! queue ! fakesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstJpegDec:jpegdec0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
0:00:00.342476843  7009   0x55c1a30400 WARN          v4l2bufferpool gstv4l2bufferpool.c:814:gst_v4l2_buffer_pool_start:<v4l2src0:pool0:src> Uncertain or not enough buffers, enabling copy threshold
/GstPipeline:pipeline0/GstJpegDec:jpegdec0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)1:4:0:0, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstQueue:queue0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)1:4:0:0, framerate=(fraction)30/1
[... cut ...]
/GstPipeline:pipeline0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)1:4:0:0, framerate=(fraction)30/1
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:03.669727283
Setting pipeline to NULL ...
Freeing pipeline ...
root@firefly:~# 
```

NIKovachev, 2023-08-03:
Hi, did you manage to reproduce, do you need any other information? I can provide test environment if needed.

JeffyCN, 2023-08-03:
0:00:00.741355943  6497   0x55bee02400 ERROR                 mppdec gstmppdec.c:1012:gst_mpp_dec_handle_frame:<mppjpegdec0> failed to send packet

It looks like the MPP doesn't accept new packet. So:
1/ did you see this error(failed to send packet) when it failed at the first time?
2/ maybe try to increase the input timeout:
gstreamer-rockchip/gst/rockchipmpp# ag MPP_INPUT_TIMEOUT_MS
gstmppdec.c
42:#define MPP_INPUT_TIMEOUT_MS 2000       /* Block timeout for MPP input queue */

NIKovachev, 2023-08-03:
The error when it failed for first time you can find in my initial comment, under "Initial error".

I have doubts there is a problem with the send packages as in the same time I can run another pipeline but with jpegdec without any problems. You can find that under "Running the same pipe but with jpegdec (works fine)".

Restart is fixing the problem.

NIKovachev, 2023-08-03:
can I pass MPP_INPUT_TIMEOUT_MS value to gst-launch somehow?

NIKovachev, 2023-08-04:
indeed it happen to be a camera issue! Thanks for your help!

closed 2023-08-04 by NIKovachev