JeffyCN/rockchip_mirrors #36: RgaBlit(1356) RGA_BLIT fail: Invalid argument RgaBlit(1357) RGA_BLIT fail: Invalid argument

yrik, 2023-09-01:
I'm getting an error when using hardware acceleration for video scale conversion. 

This video conversion consumes 80% of one core and there are complaints that this computer is too slow.
I run it on Rock 5b with 8GB mem. 

Gstreamer is built from the latest `radxa/meta-rockchip` yocto configuration.

Is there any way to be able to use hardware acceleration for video conversion/scaling?

```
root@rockchip-rk3588-rock-5b:~#  GST_VIDEO_CONVERT_USE_RGA=1 gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw, width=1920, height=1080, format=BGR, framerate=30/1, colorimetry=1:4:7:1, interlace-mode=progressive ! videoconvert ! videoscale ! video/x-raw,width=640,height=360,format=BGR ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '6'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
rga_api version 1.3.2_[0] (RGA is compiling with meson base: $PRODUCT_BASE)
 RgaBlit(1356) RGA_BLIT fail: Invalid argument RgaBlit(1357) RGA_BLIT fail: Invalid argument
fd-vir-phy-hnd-format[0, 0x7f8c1c8000, (nil), (nil), 0]
rect[0, 0, 1920, 1080, 1920, 1080, 1792, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[0, 0x7f78038090, (nil), (nil), 0]
rect[0, 0, 640, 360, 640, 360, 1792, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
WARNING: from element /GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0/GstDfbVideoSink:autovideosink0-actual-sink-dfbvideo: A lot of buffers are being dropped.
Additional debug info:
../gstreamer-1.20.6/libs/gst/base/gstbasesink.c(3143): gst_base_sink_is_too_late (): /GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0/GstDfbVideoSink:autovideosink0-actual-sink-dfbvideo:
There may be a timestamping problem, or this computer is too slow.
 RgaBlit(1356) RGA_BLIT fail: Invalid argument RgaBlit(1357) RGA_BLIT fail: Invalid argument
fd-vir-phy-hnd-format[0, 0x7f8c1c8000, (nil), (nil), 0]
rect[0, 0, 1920, 1080, 1920, 1080, 1792, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[0, 0x7f6c04c100, (nil), (nil), 0]
rect[0, 0, 640, 360, 640, 360, 1792, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
```

yrik, 2023-09-01:
Also, this is how a simple test output looks like.
It supposed to be blocks with solid colors, without horizontal lines and with bright colors. Instead there are some extra lines and missing colors.

```
gst-launch-1.0 videotestsrc ! videoconvert ! video/x-raw,width=1280,height=720,format=RGB ! autovideosink
```

[image]

JeffyCN, 2023-09-11:
sorry, i don't know much about RGA :(

the RGA changes a lot in RK3588 (hardware + kernel driver + userspace lib), it's too hard for me to follow those changes. please mail the RGA maintaner(Yu Qiaowei <cerf.yu@rock-chips.com>) for more details, i've heard it needs special version of kernel driver and userspace lib to work.

JeffyCN, 2023-09-11:
there're lots of branches for librga, i've uploaded some here:
https://github.com/JeffyCN/mirrors/branches

maybe the "multi" branch would work for rk3588

closed 2023-09-26 by JeffyCN