JeffyCN/rockchip_mirrors #39: gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame / can't process this frame

NIKovachev, 2023-12-05:
Hi,
 I cannot get RGB format from mppjpegdec. I had an older version shipped with the debian gst that was working fine for all formats apart from RGB. For the RGB issue then I created isseu #31 . The same was marked as solved but now when I tested it even the RGBA is not working. 

RGB error:
```
root@firefly:~/mirrors-gstreamer-rockchip/builddir# gst-launch-1.0 -v --gst-debug=2 shmsrc socket-path=/var/ramfs/camera_shm ! image/jpeg,width=3840,height=2160,framerate=20/1 ! mppjpegdec format=BGR dma-feature=1 max-errors=-1 ! fpsdisplaysink video-sink=fakesink text-overlay=false -ve
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)20/1, parsed=(boolean)true
0:00:00.062638634 19674   0x559f432c60 WARN              video-info video-info.c:196:validate_colorimetry: color matrix RGB is only supported with RGB format, NV12 is not
0:00:00.062665759 19674   0x559f432c60 WARN              video-info video-info.c:520:gst_video_info_from_caps: invalid colorimetry, using default
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)BGR, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw(memory:DMABuf), format=(string)BGR, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)BGR, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)BGR, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)20/1, parsed=(boolean)true
0:00:00.062956842 19674   0x559f432c60 WARN            videodecoder gstvideodecoder.c:2573:gst_video_decoder_chain:<mppjpegdec0> Received buffer without a new-segment. Assuming timestamps start from 0.
0:00:00.073716567 19674   0x559f433180 WARN                  mppdec gstmppdec.c:591:gst_mpp_dec_get_frame:<mppjpegdec0> MPP is not able to generate pts
0:00:00.074007942 19674   0x559f433180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.074093109 19674   0x559f433180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.133421321 19674   0x559f433180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.133531280 19674   0x559f433180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.196971665 19674   0x559f433180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.197036707 19674   0x559f433180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.254252374 19674   0x559f433180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame

```

RGBA error:
```
root@firefly:~/mirrors-gstreamer-rockchip/builddir# gst-launch-1.0 -v --gst-debug=2 shmsrc socket-path=/var/ramfs/camera_shm ! image/jpeg,width=3840,height=2160,framerate=20/1 ! mppjpegdec format=BGR565 dma-feature=1 max-errors=-1 ! fpsdisplaysink video-sink=fakesink text-overlay=false -ve
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)20/1, parsed=(boolean)true
0:00:00.060338336 19830   0x55a3276060 ERROR             mppjpegdec gstmppjpegdec.c:281:gst_mpp_jpeg_dec_set_format:<mppjpegdec0> unsupported video format
0:00:00.060379169 19830   0x55a3276060 WARN            videodecoder gstvideodecoder.c:811:gst_video_decoder_setcaps:<mppjpegdec0> Subclass refused caps
0:00:00.060823670 19830   0x55a3276060 ERROR             mppjpegdec gstmppjpegdec.c:281:gst_mpp_jpeg_dec_set_format:<mppjpegdec0> unsupported video format
0:00:00.060850503 19830   0x55a3276060 WARN            videodecoder gstvideodecoder.c:811:gst_video_decoder_setcaps:<mppjpegdec0> Subclass refused caps
0:00:00.061242504 19830   0x55a3276060 ERROR             mppjpegdec gstmppjpegdec.c:281:gst_mpp_jpeg_dec_set_format:<mppjpegdec0> unsupported video format
0:00:00.061267295 19830   0x55a3276060 WARN            videodecoder gstvideodecoder.c:811:gst_video_decoder_setcaps:<mppjpegdec0> Subclass refused caps
0:00:00.061321254 19830   0x55a3276060 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<shmsrc0> error: Internal data stream error.
0:00:00.061337587 19830   0x55a3276060 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<shmsrc0> error: streaming stopped, reason not-negotiated (-4)
ERROR: from element /GstPipeline:pipeline0/GstShmSrc:shmsrc0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstShmSrc:shmsrc0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
0:00:00.061796088 19830   0x55a3276060 ERROR             mppjpegdec gstmppjpegdec.c:281:gst_mpp_jpeg_dec_set_format:<mppjpegdec0> unsupported video format
0:00:00.061819129 19830   0x55a3276060 WARN            videodecoder gstvideodecoder.c:811:gst_video_decoder_setcaps:<mppjpegdec0> Subclass refused caps
0:00:00.062001130 19830   0x55a3276060 ERROR             mppjpegdec gstmppjpegdec.c:281:gst_mpp_jpeg_dec_set_format:<mppjpegdec0> unsupported video format
0:00:00.062017171 19830   0x55a3276060 WARN            videodecoder gstvideodecoder.c:811:gst_video_decoder_setcaps:<mppjpegdec0> Subclass refused caps
Freeing pipeline ...
root@firefly:~/mirrors-gstreamer-rockchip/builddir# 
root@firefly:~/mirrors-gstreamer-rockchip/builddir# 
root@firefly:~/mirrors-gstreamer-rockchip/builddir# 
root@firefly:~/mirrors-gstreamer-rockchip/builddir# gst-launch-1.0 -v --gst-debug=2 shmsrc socket-path=/var/ramfs/camera_shm ! image/jpeg,width=3840,height=2160,framerate=20/1 ! mppjpegdec format=RGBA dma-feature=1 max-errors=-1 ! fpsdisplaysink video-sink=fakesink text-overlay=false -ve
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)20/1, parsed=(boolean)true
0:00:00.116969648 20414   0x5583f1ec60 ERROR             mppjpegdec gstmppjpegdec.c:281:gst_mpp_jpeg_dec_set_format:<mppjpegdec0> unsupported video format
[... cut ...]
0:00:00.119783069 20414   0x5583f1ec60 WARN            videodecoder gstvideodecoder.c:811:gst_video_decoder_setcaps:<mppjpegdec0> Subclass refused caps
0:00:00.120452153 20414   0x5583f1ec60 ERROR             mppjpegdec gstmppjpegdec.c:281:gst_mpp_jpeg_dec_set_format:<mppjpegdec0> unsupported video format
0:00:00.120509903 20414   0x5583f1ec60 WARN            videodecoder gstvideodecoder.c:811:gst_video_decoder_setcaps:<mppjpegdec0> Subclass refused caps
Freeing pipeline ...
root@firefly:~/mirrors-gstreamer-rockchip/builddir# 

```

JeffyCN, 2023-12-05:
it looks like something wrong in your rga setup

NIKovachev, 2023-12-05:
Maybe I did something wrong? I downloaded the latest version of the mirrors-gstreamer-rockchip, compiled and  installed to the latest gst provided by the debian repo. Do I need to do something else? 

```
root@firefly:~/mirrors-gstreamer-rockchip/builddir# meson install
ninja: Entering directory `/root/mirrors-gstreamer-rockchip/builddir'
ninja: no work to do.
Installing gst/rkximage/libgstrkximage.so to /usr/local/lib/aarch64-linux-gnu/gstreamer-1.0
Installing gst/rockchipmpp/libgstrockchipmpp.so to /usr/local/lib/aarch64-linux-gnu/gstreamer-1.0
Installing gst/kmssrc/libgstkmssrc.so to /usr/local/lib/aarch64-linux-gnu/gstreamer-1.0
root@firefly:~/mirrors-gstreamer-rockchip/builddir# 

```

JeffyCN, 2023-12-05:
if you are compiling it by yourself, maybe should install librga*-dev

NIKovachev, 2023-12-05:
I have it installed. this is the whole meson output. I really appreciate your help!

```
root@firefly:~/mirrors-gstreamer-rockchip/builddir# apt search librga*-dev
Sorting... Done
Full Text Search... Done
librga-dev/now 2.2.0-1 arm64 [installed,local]
  Userspace interface to Rockchip RGA 2D accelerator

root@firefly:~/mirrors-gstreamer-rockchip/builddir# 

root@firefly:~/mirrors-gstreamer-rockchip# meson setup builddir && cd builddir
The Meson build system
Version: 1.3.0
Source dir: /root/mirrors-gstreamer-rockchip
Build dir: /root/mirrors-gstreamer-rockchip/builddir
Build type: native build
Project name: gst-plugins-bad
Project version: 1.14.4

C compiler for the host machine: cc (gcc 10.2.1 "cc (Debian 10.2.1-6) 10.2.1 20210110")
C linker for the host machine: cc ld.bfd 2.35.2
Host machine cpu family: aarch64
Host machine cpu: aarch64
Compiler for C supports arguments -fvisibility=hidden: YES 
Compiler for C supports arguments -fno-strict-aliasing: YES 
Message: Disabling GLib cast checks
Message: Disabling deprecated GLib API
Found pkg-config: YES (/usr/bin/pkg-config) 0.29.2
Run-time dependency gstreamer-1.0 found: YES 1.18.5
Run-time dependency gstreamer-base-1.0 found: YES 1.18.5
Run-time dependency gstreamer-allocators-1.0 found: YES 1.18.5
Run-time dependency gstreamer-video-1.0 found: YES 1.18.5
Run-time dependency gstreamer-pbutils-1.0 found: YES 1.18.5
Run-time dependency glib-2.0 found: YES 2.66.8
Run-time dependency x11 found: YES 1.7.2
Run-time dependency libdrm found: YES 2.4.104
Run-time dependency rockchip_mpp found: YES 1.3.8
Run-time dependency librga found: YES 2.1.0
meson.build:92: DEPRECATION: configuration_data.set10 with number. the `set10` method should only be used with booleans
Header "gst/video/video-format.h" has symbol "GST_VIDEO_FORMAT_NV12_10LE40" with dependency gstreamer-video-1.0: YES 
meson.build:107: DEPRECATION: configuration_data.set10 with number. the `set10` method should only be used with booleans
Header "gst/video/video-format.h" has symbol "GST_VIDEO_FORMAT_NV16_10LE40" with dependency gstreamer-video-1.0: NO 
Checking for function "drmModeGetFB2" with dependency libdrm: YES 
Checking for function "drmCloseBufferHandle" with dependency libdrm: NO 
Configuring config.h using configuration
Program python3 found: YES (/usr/bin/python3)
WARNING: You should add the boolean check kwarg to the run_command call.
         It currently defaults to false,
         but it will default to true in future releases of meson.
         See also: https://github.com/mesonbuild/meson/issues/9300
Build targets in project: 3

Found ninja-1.10.1 at /usr/bin/ninja
root@firefly:~/mirrors-gstreamer-rockchip/builddir# 
root@firefly:~/mirrors-gstreamer-rockchip/builddir# meson compile
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja
[1/18] Compiling C object gst/rkximage/libgstrkximage.so.p/gstkmsbufferpool.c.o
../gst/rkximage/gstkmsbufferpool.c: In function ‘gst_kms_buffer_pool_get_options’:
../gst/rkximage/gstkmsbufferpool.c:53:50: warning: unused parameter ‘pool’ [-Wunused-parameter]
   53 | gst_kms_buffer_pool_get_options (GstBufferPool * pool)
      |                                  ~~~~~~~~~~~~~~~~^~~~
../gst/rkximage/gstkmsbufferpool.c: In function ‘gst_kms_buffer_pool_alloc_buffer’:
../gst/rkximage/gstkmsbufferpool.c:134:34: warning: unused parameter ‘params’ [-Wunused-parameter]
  134 |     GstBufferPoolAcquireParams * params)
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
[5/18] Compiling C object gst/rkximage/libgstrkximage.so.p/gstkmsutils.c.o
../gst/rkximage/gstkmsutils.c: In function ‘gst_video_format_from_drm’:
../gst/rkximage/gstkmsutils.c:82:17: warning: comparison of integer expressions of different signedness: ‘gint’ {aka ‘int’} and ‘long unsigned int’ [-Wsign-compare]
[... 71 lines cut ...]
../gst/rkximage/ximagesink.c: In function ‘drm_ensure_allowed_caps’:
../gst/rkximage/ximagesink.c:446:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  446 |   for (i = 0; i < plane->count_formats; i++) {
      |                 ^
../gst/rkximage/ximagesink.c: In function ‘drm_plane_set_property’:
../gst/rkximage/ximagesink.c:510:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  510 |   for (i = 0; i < props->count_props; i++) {
      |                 ^
../gst/rkximage/ximagesink.c: In function ‘sync_handler’:
../gst/rkximage/ximagesink.c:884:20: warning: unused parameter ‘fd’ [-Wunused-parameter]
  884 | sync_handler (gint fd, guint frame, guint sec, guint usec, gpointer data)
      |               ~~~~~^~
../gst/rkximage/ximagesink.c:884:30: warning: unused parameter ‘frame’ [-Wunused-parameter]
  884 | sync_handler (gint fd, guint frame, guint sec, guint usec, gpointer data)
      |                        ~~~~~~^~~~~
../gst/rkximage/ximagesink.c:884:43: warning: unused parameter ‘sec’ [-Wunused-parameter]
  884 | sync_handler (gint fd, guint frame, guint sec, guint usec, gpointer data)
      |                                     ~~~~~~^~~
../gst/rkximage/ximagesink.c:884:54: warning: unused parameter ‘usec’ [-Wunused-parameter]
  884 | sync_handler (gint fd, guint frame, guint sec, guint usec, gpointer data)
      |                                                ~~~~~~^~~~
[18/18] Linking target gst/rkximage/libgstrkximage.so
root@firefly:~/mirrors-gstreamer-rockchip/builddir# 

```

JeffyCN, 2023-12-05:
well, check the log again, "unsupported video format" it means the original decode format is not supported.

this logic is in gst_mpp_jpeg_dec_set_format, it parse the src_format from gst's input cap and filter out unsupported format.

did your setup environment changed a lot? i don't understand why it worked at before

JeffyCN, 2023-12-05:
maybe try to add a jpegparser before the jpegdec, maybe it's because the cap doesn't contain the original format

NIKovachev, 2023-12-05:
I tried to reinstall the linux image, just to be sure it's not messed up. I'm using the stock "AIO-3588Q-EDP_Debian11-r106b_v1.0.6c_230114". The only difference is the mirrors-gstreamer-rockchip. After the meson compile/install I copy the following files manualy:
```
cp gst/rkximage/libgstrkximage.so to /usr/local/lib/gstreamer-1.0
cp gst/rockchipmpp/libgstrockchipmpp.so to /usr/local/lib/gstreamer-1.0
cp gst/kmssrc/libgstkmssrc.so to /usr/local/lib/gstreamer-1.0
```

Unfortunately the output is still the same:

```
root@firefly:~/mirrors-gstreamer-rockchip/builddir# gst-launch-1.0 -v --gst-debug=2 shmsrc socket-path=/var/ramfs/camera_shm ! image/jpeg,width=3840,height=2160,framerate=20/1 ! jpegparse ! mppjpegdec format=RGB dma-feature=1 max-errors=-1 ! fpsdisplaysink video-sink="shmsink socket-path=/var/ramfs/camera_shm_rgb sync=true async=false wait-for-connection=false shm-size=100000000" text-overlay=false -ve
Setting pipeline to PAUSED ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstShmSink:shmsink0: sync = true
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstShmSink:shmsink0: sync = true
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstJpegParse:jpegparse0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstJpegParse:jpegparse0.GstPad:src: caps = image/jpeg, parsed=(boolean)true, format=(string)UYVY, width=(int)3840, height=(int)2160, framerate=(fraction)20/1
0:00:00.076721811 74032   0x5582aad5e0 WARN              video-info video-info.c:196:validate_colorimetry: color matrix RGB is only supported with RGB format, NV12 is not
0:00:00.076788894 74032   0x5582aad5e0 WARN              video-info video-info.c:520:gst_video_info_from_caps: invalid colorimetry, using default
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)RGB, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw(memory:DMABuf), format=(string)RGB, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstShmSink:shmsink0.GstPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)RGB, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)RGB, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:sink: caps = image/jpeg, parsed=(boolean)true, format=(string)UYVY, width=(int)3840, height=(int)2160, framerate=(fraction)20/1
0:00:00.135118524 74032   0x5582ab0180 WARN                  mppdec gstmppdec.c:591:gst_mpp_dec_get_frame:<mppjpegdec0> MPP is not able to generate pts
0:00:00.135330274 74032   0x5582ab0180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.135377816 74032   0x5582ab0180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.172378721 74032   0x5582ab0180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.172437929 74032   0x5582ab0180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.229863225 74032   0x5582ab0180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.229949558 74032   0x5582ab0180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
0:00:00.292788317 74032   0x5582ab0180 WARN                  mppdec gstmppdec.c:821:gst_mpp_dec_get_gst_buffer:<mppjpegdec0> unable to convert frame
0:00:00.292876400 74032   0x5582ab0180 WARN                  mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppjpegdec0> can't process this frame
[... cut ...]
0:00:00.422017253 74032   0x55827fef60 WARN                 shmsink gstshmsink.c:840:pollthread_func:<shmsink0> error: gst_poll_wait returned -1, errno: 16
ERROR: from element /GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstShmSink:shmsink0: Failed waiting on fd activity
Additional debug info:
../sys/shm/gstshmsink.c(840): pollthread_func (): /GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstShmSink:shmsink0:
gst_poll_wait returned -1, errno: 16
An error happened while waiting for EOS
Freeing pipeline ...
root@firefly:~/mirrors-gstreamer-rockchip/builddir# 

```

JeffyCN, 2023-12-05:
the rgba works again right?(it would use mpp's internal postproc to convert)

the bgr is not supported by mpp(gst_mpp_jpeg_dec_pp_formats), so it would use rga instead, so maybe check is there any rga related error log in kernel dmesg

or capture all logs with GST_DEBUG=mpp*:5 and upload it here

NIKovachev, 2023-12-05:
Yes ARGB is working fine now! There is an error indeed. The gst.out with gst-debug=5  is attached. The GST_DEBUG=mpp*:5  output is below as well.

```
root@firefly:~# gst-launch-1.0 -v --gst-debug=5 shmsrc socket-path=/var/ramfs/camera_shm ! image/jpeg,width=3840,height=2160,framerate=20/1 ! mppjpegdec format=RGB dma-feature=1 max-errors=-1 ! fpsdisplaysink video-sink="shmsink socket-path=/var/ramfs/camera_shm_rgb sync=true async=false wait-for-connection=false shm-size=100000000" text-overlay=false -ve &> gst.out 
^Croot@firefly:~tail /var/log/messages 
Dec  5 18:49:59 firefly mpp[15020]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
Dec  5 18:49:59 firefly mpp[15020]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
Dec  5 18:49:59 firefly mpp[15020]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
Dec  5 18:49:59 firefly mpp[15020]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
Dec  5 18:49:59 firefly mpp[15020]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
Dec  5 18:50:44 firefly mpp[15183]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
Dec  5 18:50:44 firefly mpp[15183]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
Dec  5 18:50:44 firefly mpp[15183]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
Dec  5 18:50:44 firefly mpp[15183]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
Dec  5 18:50:44 firefly mpp[15183]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
root@firefly:~# date
Tue Dec  5 18:50:53 UTC 2023
root@firefly:~# 
```

[
[gst.zip](https://github.com/JeffyCN/rockchip_mirrors/files/13569727/gst.zip)
](url)

The GST_DEBUG=mpp*:5 

```
root@firefly:~# gst-launch-1.0 -v --gst-debug=2 shmsrc socket-path=/var/ramfs/camera_shm ! image/jpeg,width=3840,height=2160,framerate=20/1 ! mppjpegdec format=RGB dma-feature=1 max-errors=-1 ! fpsdisplaysink video-sink="shmsink socket-path=/var/ramfs/camera_shm_rgb sync=true async=false wait-for-connection=false shm-size=100000000" text-overlay=false -ve 
Setting pipeline to PAUSED ...
0:00:00.058264531 15953   0x558cb21d80 DEBUG             mppjpegdec gstmppjpegdec.c:181:gst_mpp_jpeg_dec_start:<mppjpegdec0> starting
0:00:00.058353489 15953   0x558cb21d80 DEBUG                 mppdec gstmppdec.c:247:gst_mpp_dec_start:<mppjpegdec0> starting
0:00:00.058405406 15953   0x558cb21d80 DEBUG                 mppdec gstmppdec.c:271:gst_mpp_dec_start:<mppjpegdec0> started
0:00:00.058412989 15953   0x558cb21d80 DEBUG             mppjpegdec gstmppjpegdec.c:202:gst_mpp_jpeg_dec_start:<mppjpegdec0> started
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstShmSink:shmsink0: sync = true
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.059091989 15953   0x558cb318c0 DEBUG                 mppdec gstmppdec.c:326:gst_mpp_dec_drain:<mppjpegdec0> draining
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstShmSink:shmsink0: sync = true
0:00:00.059139239 15953   0x558cb318c0 DEBUG                 mppdec gstmppdec.c:225:gst_mpp_dec_reset:<mppjpegdec0> resetting
0:00:00.095072883 15953   0x558cb318c0 DEBUG                 mppdec gstmppdec.c:348:gst_mpp_dec_set_format:<mppjpegdec0> setting format: image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)20/1, parsed=(boolean)true
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)20/1, parsed=(boolean)true
0:00:00.097000217 15953   0x558cb318c0 DEBUG                    mpp gstmpp.c:344:gst_mpp_video_info_update_format: update format to RGB (3840x2160)
0:00:00.097291593 15953   0x558cb318c0 INFO              mppjpegdec gstmppjpegdec.c:296:gst_mpp_jpeg_dec_set_format:<mppjpegdec0> convert from NV12 (3840x2160) to RGB (3840x2160)
0:00:00.097340593 15953   0x558cb318c0 DEBUG                    mpp gstmpp.c:344:gst_mpp_video_info_update_format: update format to NV12 (3840x2160)
0:00:00.097477384 15953   0x558cb318c0 WARN              video-info video-info.c:196:validate_colorimetry: color matrix RGB is only supported with RGB format, NV12 is not
0:00:00.097517634 15953   0x558cb318c0 WARN              video-info video-info.c:520:gst_video_info_from_caps: invalid colorimetry, using default
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)RGB, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)sRGB, framerate=(fraction)20/1
0:00:00.098501260 15953   0x558cb318c0 WARN            videodecoder gstvideodecoder.c:2573:gst_video_decoder_chain:<mppjpegdec0> Received buffer without a new-segment. Assuming timestamps start from 0.
[... cut ...]
Dec  5 18:54:26 firefly mpp[15953]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
Dec  5 18:54:26 firefly mpp[15953]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
Dec  5 18:54:26 firefly mpp[15953]: mpp_info: mpp version: 2edfc6db author: Chandler Chen 2022-11-22 [vepu540c_hevc]: fix argb8888 encode error
root@firefly:~# date
Tue Dec  5 18:54:30 UTC 2023
root@firefly:~# 
```

JeffyCN, 2023-12-06:
it looks like rga is disabled, maybe check the env(run export), is there something like GST_MPP_NO_RGA=1? is so, try to unset it

NIKovachev, 2023-12-06:
That fixed the issue!! I have no idea how that variable was set. I really appreciate your support, thank you!

For future reference GST_MPP_NO_RGA was 1, I set it to 0 and it works:
```
root@firefly:~# env|grep GST
GST_V4L2_PREFERRED_FOURCC=NV12:YU12:NV16:YUY2
GST_VIDEO_CONVERT_PREFERRED_FORMAT=NV12:NV16:I420:YUY2
GST_MPP_NO_RGA=0
GST_GL_PLATFORM=egl
GST_V4L2_USE_LIBV4L2=1
GST_INSPECT_NO_COLORS=1
GST_DEBUG_NO_COLOR=1
GST_V4L2SRC_DEFAULT_DEVICE=/dev/video-camera0
GST_V4L2SRC_RK_DEVICES=_mainpath:_selfpath:_bypass:_scale
GST_GL_API=gles2
GST_V4L2SRC_MAX_RESOLUTION=3840x2160
GST_MPP_VIDEODEC_DEFAULT_ARM_AFBC=1
root@firefly:~# 

```

closed 2023-12-06 by NIKovachev