JeffyCN/rockchip_mirrors #40: mpph264enc hand and doesn't output anything

NIKovachev, 2023-12-19:
Hello,
I cannot get any output from mpph264enc. I tried all kind of combinations but no luck.

```
(base) root@firefly:~# gst-launch-1.0 --gst-debug=4 videotestsrc ! 'video/x-raw,format=YUY2,width=1920,height=1080,framrate=30/1' ! mpph264enc ! 'video/x-h264' ! h264parse ! 'video/x-h264' ! fpsdisplaysink video-sink=fakesink text-overlay=false sync=false -ve
0:00:00.001258252 982791   0x558a0b42c0 INFO                GST_INIT gstmessage.c:129:_priv_gst_message_initialize: init messages
0:00:00.002497713 982791   0x558a0b42c0 INFO                GST_INIT gstcontext.c:85:_priv_gst_context_initialize: init contexts
0:00:00.003057131 982791   0x558a0b42c0 INFO      GST_PLUGIN_LOADING gstplugin.c:325:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.003313798 982791   0x558a0b42c0 INFO      GST_PLUGIN_LOADING gstplugin.c:233:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.003354339 982791   0x558a0b42c0 INFO      GST_PLUGIN_LOADING gstplugin.c:235:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.003389339 982791   0x558a0b42c0 INFO            GST_REGISTRY gstregistry.c:1755:ensure_current_registry: reading registry cache: /root/.cache/gstreamer-1.0/registry.aarch64.bin
0:00:00.042374369 982791   0x558a0b42c0 INFO            GST_REGISTRY gstregistrybinary.c:666:priv_gst_registry_binary_read_cache: loaded /root/.cache/gstreamer-1.0/registry.aarch64.bin in 0.038918 seconds
0:00:00.042504453 982791   0x558a0b42c0 INFO            GST_REGISTRY gstregistry.c:1614:scan_and_update_registry: Validating plugins from registry cache: /root/.cache/gstreamer-1.0/registry.aarch64.bin
0:00:00.044417498 982791   0x558a0b42c0 INFO            GST_REGISTRY gstregistry.c:1713:scan_and_update_registry: Registry cache has not changed
0:00:00.044437623 982791   0x558a0b42c0 INFO            GST_REGISTRY gstregistry.c:1790:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.044448707 982791   0x558a0b42c0 INFO                GST_INIT gst.c:806:init_post: GLib runtime version: 2.66.8
0:00:00.044453957 982791   0x558a0b42c0 INFO                GST_INIT gst.c:808:init_post: GLib headers version: 2.66.8
0:00:00.044457457 982791   0x558a0b42c0 INFO                GST_INIT gst.c:810:init_post: initialized GStreamer successfully
0:00:00.044525998 982791   0x558a0b42c0 INFO            GST_PIPELINE gstparse.c:344:gst_parse_launch_full: parsing pipeline description 'videotestsrc ! video/x-raw,format=YUY2,width=1920,height=1080,framrate=30/1 ! mpph264enc ! video/x-h264 ! h264parse ! video/x-h264 ! fpsdisplaysink video-sink=fakesink text-overlay=false sync=false '
0:00:00.048696256 982791   0x558a0b42c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstvideotestsrc.so" loaded
0:00:00.048722214 982791   0x558a0b42c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "videotestsrc"
0:00:00.048952632 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseSrc@0x558a3314e0> adding pad 'src'
0:00:00.051724928 982791   0x558a0b42c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrockchipmpp.so" loaded
0:00:00.051743595 982791   0x558a0b42c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "mpph264enc"
0:00:00.051926762 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstVideoEncoder@0x558a33d0c0> adding pad 'sink'
0:00:00.051942512 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstVideoEncoder@0x558a33d0c0> adding pad 'src'
0:00:00.053366431 982791   0x558a0b42c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstvideoparsersbad.so" loaded
0:00:00.053380723 982791   0x558a0b42c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "h264parse"
0:00:00.053466182 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseParse@0x558a341280> adding pad 'sink'
0:00:00.053482223 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseParse@0x558a341280> adding pad 'src'
0:00:00.053507890 982791   0x558a0b42c0 INFO               baseparse gstbaseparse.c:4041:gst_base_parse_set_pts_interpolation:<GstH264Parse@0x558a341280> PTS interpolation: no
0:00:00.053515473 982791   0x558a0b42c0 INFO               baseparse gstbaseparse.c:4059:gst_base_parse_set_infer_ts:<GstH264Parse@0x558a341280> TS inferring: no
0:00:00.055993186 982791   0x558a0b42c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstdebugutilsbad.so" loaded
0:00:00.056008061 982791   0x558a0b42c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "fpsdisplaysink"
0:00:00.056134936 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstFPSDisplaySink@0x558a34c150> adding pad 'sink'
0:00:00.056163520 982791   0x558a0b42c0 INFO            GST_PIPELINE gstparse.c:344:gst_parse_launch_full: parsing pipeline description 'fakesink'
0:00:00.056738104 982791   0x558a0b42c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstcoreelements.so" loaded
0:00:00.056753854 982791   0x558a0b42c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "fakesink"
0:00:00.056879563 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseSink@0x558a3582d0> adding pad 'sink'
0:00:00.056956563 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstelement.c:1009:gst_element_get_static_pad: found pad fakesink0:sink
0:00:00.056985730 982791   0x558a0b42c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "pipeline"
0:00:00.057078771 982791   0x558a0b42c0 INFO            GST_PIPELINE gst/parse/grammar.y:689:gst_parse_perform_link: linking some pad of GstVideoTestSrc named videotestsrc0 to some pad of GstMppH264Enc named mpph264enc0 (0/0) with caps "video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, framrate=(fraction)30/1"
0:00:00.057089563 982791   0x558a0b42c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "capsfilter"
0:00:00.057157522 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseTransform@0x558a35c170> adding pad 'sink'
0:00:00.057170355 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseTransform@0x558a35c170> adding pad 'src'
0:00:00.057181438 982791   0x558a0b42c0 INFO              GST_STATES gstbin.c:2115:gst_bin_get_state_func:<pipeline0> getting state
0:00:00.057200397 982791   0x558a0b42c0 INFO              GST_STATES gstelement.c:2769:gst_element_continue_state:<capsfilter0> completed state change to NULL
0:00:00.057217605 982791   0x558a0b42c0 INFO               GST_EVENT gstevent.c:1610:gst_event_new_reconfigure: creating reconfigure event
0:00:00.057230730 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstutils.c:1816:gst_element_link_pads_full: trying to link element videotestsrc0:(any) to element capsfilter0:sink
0:00:00.057238022 982791   0x558a0b42c0 INFO        GST_ELEMENT_PADS gstelement.c:1009:gst_element_get_static_pad: found pad capsfilter0:sink
0:00:00.057244147 982791   0x558a0b42c0 INFO                GST_PADS gstutils.c:1632:prepare_link_maybe_ghosting: videotestsrc0 and capsfilter0 in same bin, no need for ghost pads
0:00:00.057263397 982791   0x558a0b42c0 INFO                GST_PADS gstpad.c:2404:gst_pad_link_prepare: trying to link videotestsrc0:src and capsfilter0:sink
0:00:00.057285272 982791   0x558a0b42c0 INFO                GST_PADS gstpad.c:4336:gst_pad_peer_query:<capsfilter0:src> pad has no peer
0:00:00.057318814 982791   0x558a0b42c0 INFO                GST_PADS gstpad.c:2612:gst_pad_link_full: linked videotestsrc0:src and capsfilter0:sink, successful
[... cut ...]
EOS on shutdown enabled -- Forcing EOS on the pipeline
^C
(base) root@firefly:~# env|grep GST
GST_V4L2_PREFERRED_FOURCC=NV12:YU12:NV16:YUY2
GST_VIDEO_CONVERT_PREFERRED_FORMAT=NV12:NV16:I420:YUY2
GST_MPP_NO_RGA=0
GST_GL_PLATFORM=egl
GST_V4L2_USE_LIBV4L2=1
GST_INSPECT_NO_COLORS=1
GST_DEBUG_NO_COLOR=1
GST_V4L2SRC_DEFAULT_DEVICE=/dev/video-camera0
GST_V4L2SRC_RK_DEVICES=_mainpath:_selfpath:_bypass:_scale
GST_GL_API=gles2
GST_V4L2SRC_MAX_RESOLUTION=3840x2160
GST_MPP_VIDEODEC_DEFAULT_ARM_AFBC=1
(base) root@firefly:~# 

```

JeffyCN, 2023-12-19:
try nv12 1920x1088

NIKovachev, 2023-12-20:
it still doesn't work with nv12 1920x1088. x264enc works fine but it consumes a lot fo CPU. 

```

(base) root@firefly:~# gst-launch-1.0 --gst-debug=4 videotestsrc ! 'video/x-raw,format=NV12,width=1920,height=1088,framrate=30/1' ! mpph264enc ! 'video/x-h264' ! h264parse ! 'video/x-h264' ! fpsdisplaysink video-sink=fakesink text-overlay=false sync=false -ve
0:00:00.000768251 113705   0x55abaf72c0 INFO                GST_INIT gstmessage.c:129:_priv_gst_message_initialize: init messages
0:00:00.001883002 113705   0x55abaf72c0 INFO                GST_INIT gstcontext.c:85:_priv_gst_context_initialize: init contexts
0:00:00.002555294 113705   0x55abaf72c0 INFO      GST_PLUGIN_LOADING gstplugin.c:325:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.002800295 113705   0x55abaf72c0 INFO      GST_PLUGIN_LOADING gstplugin.c:233:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.002843170 113705   0x55abaf72c0 INFO      GST_PLUGIN_LOADING gstplugin.c:235:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.002876128 113705   0x55abaf72c0 INFO            GST_REGISTRY gstregistry.c:1755:ensure_current_registry: reading registry cache: /root/.cache/gstreamer-1.0/registry.aarch64.bin
0:00:00.039159789 113705   0x55abaf72c0 INFO            GST_REGISTRY gstregistrybinary.c:666:priv_gst_registry_binary_read_cache: loaded /root/.cache/gstreamer-1.0/registry.aarch64.bin in 0.036228 seconds
0:00:00.039269747 113705   0x55abaf72c0 INFO            GST_REGISTRY gstregistry.c:1614:scan_and_update_registry: Validating plugins from registry cache: /root/.cache/gstreamer-1.0/registry.aarch64.bin
0:00:00.041217499 113705   0x55abaf72c0 INFO            GST_REGISTRY gstregistry.c:1713:scan_and_update_registry: Registry cache has not changed
0:00:00.041237624 113705   0x55abaf72c0 INFO            GST_REGISTRY gstregistry.c:1790:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.041244916 113705   0x55abaf72c0 INFO                GST_INIT gst.c:806:init_post: GLib runtime version: 2.66.8
0:00:00.041249583 113705   0x55abaf72c0 INFO                GST_INIT gst.c:808:init_post: GLib headers version: 2.66.8
0:00:00.041253666 113705   0x55abaf72c0 INFO                GST_INIT gst.c:810:init_post: initialized GStreamer successfully
0:00:00.041306458 113705   0x55abaf72c0 INFO            GST_PIPELINE gstparse.c:344:gst_parse_launch_full: parsing pipeline description 'videotestsrc ! video/x-raw,format=NV12,width=1920,height=1088,framrate=30/1 ! mpph264enc ! video/x-h264 ! h264parse ! video/x-h264 ! fpsdisplaysink video-sink=fakesink text-overlay=false sync=false '
0:00:00.050603050 113705   0x55abaf72c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstvideotestsrc.so" loaded
0:00:00.050635133 113705   0x55abaf72c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "videotestsrc"
0:00:00.050945175 113705   0x55abaf72c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseSrc@0x55abd72d10> adding pad 'src'
0:00:00.054235762 113705   0x55abaf72c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrockchipmpp.so" loaded
0:00:00.054272804 113705   0x55abaf72c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "mpph264enc"
0:00:00.054536471 113705   0x55abaf72c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstVideoEncoder@0x55abd7c040> adding pad 'sink'
0:00:00.054556304 113705   0x55abaf72c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstVideoEncoder@0x55abd7c040> adding pad 'src'
0:00:00.058953475 113705   0x55abaf72c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstvideoparsersbad.so" loaded
0:00:00.058985558 113705   0x55abaf72c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "h264parse"
0:00:00.059290059 113705   0x55abaf72c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseParse@0x55abd83f10> adding pad 'sink'
0:00:00.059320975 113705   0x55abaf72c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseParse@0x55abd83f10> adding pad 'src'
0:00:00.059363850 113705   0x55abaf72c0 INFO               baseparse gstbaseparse.c:4041:gst_base_parse_set_pts_interpolation:<GstH264Parse@0x55abd83f10> PTS interpolation: no
0:00:00.059376100 113705   0x55abaf72c0 INFO               baseparse gstbaseparse.c:4059:gst_base_parse_set_infer_ts:<GstH264Parse@0x55abd83f10> TS inferring: no
0:00:00.062600479 113705   0x55abaf72c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstdebugutilsbad.so" loaded
0:00:00.062621770 113705   0x55abaf72c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "fpsdisplaysink"
0:00:00.062940854 113705   0x55abaf72c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstFPSDisplaySink@0x55abd8e150> adding pad 'sink'
[... cut ...]
0:00:00.110112151 113705   0x55abda1580 INFO                 basesrc gstbasesrc.c:3018:gst_base_src_loop:<videotestsrc0> marking pending DISCONT
0:00:00.110411984 113705   0x55abda1580 INFO                    task gsttask.c:516:gst_task_set_lock: setting stream lock 0x55abd74620 on task 0x7f7c024830
0:00:00.110466818 113705   0x55abda1580 INFO                GST_PADS gstpad.c:6270:gst_pad_start_task:<mpph264enc0:src> created task 0x7f7c024830

^Chandling interrupt.
Interrupt: Stopping pipeline ...
EOS on shutdown enabled -- Forcing EOS on the pipeline
^C
(base) root@firefly:~# 
```

NIKovachev, 2023-12-20:
It works!! Thanks! I did the below and the issues is solved. But when I connect the whole pipeline I get a new error. I will raise a new beg for it!

```
I downloaded: https://github.com/JeffyCN/rockchip_mirrors/tree/gstreamer-rockchip
executed:
meson setup build; cd build
meson compile
cp ./gst/rockchipmpp/libgstrockchipmpp.so /usr/local/lib/gstreamer-1.0/libgstrockchipmpp.so
cp ./gst/rockchipmpp/libgstrockchipmpp.so /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrockchipmpp.so
```

closed 2023-12-20 by NIKovachev