JeffyCN/rockchip_mirrors #41: RgaBlit(1465) RGA_BLIT fail: Invalid argument

NIKovachev, 2023-12-20:
Hi,
the pipeline runs for few minutes and it then fails. with RGA_BLIT fail: Invalid argument.

```
(base) root@firefly:~# env|grep GST
GST_V4L2_PREFERRED_FOURCC=NV12:YU12:NV16:YUY2
GST_VIDEO_CONVERT_PREFERRED_FORMAT=NV12:NV16:I420:YUY2
GST_MPP_NO_RGA=0
GST_GL_PLATFORM=egl
GST_V4L2_USE_LIBV4L2=1
GST_INSPECT_NO_COLORS=1
GST_DEBUG_NO_COLOR=1
GST_V4L2SRC_DEFAULT_DEVICE=/dev/video-camera0
GST_V4L2SRC_RK_DEVICES=_mainpath:_selfpath:_bypass:_scale
GST_GL_API=gles2
GST_V4L2SRC_MAX_RESOLUTION=3840x2160
GST_MPP_VIDEODEC_DEFAULT_ARM_AFBC=1
(base) root@firefly:~/gst-rtsp-server/examples# gst-launch-1.0 --gst-debug=4 v4l2src device=/dev/video12 ! image/jpeg,width=3840,height=2160,framerate=10/1 ! mppjpegdec format=RGB dma-feature=0 max-errors=-1 ! mpph264enc ! rtph264pay name=pay0 pt=96 ! fpsdisplaysink video-sink="fakesink" text-overlay=false -ve
0:00:00.000794793 143596   0x55702b92c0 INFO                GST_INIT gstmessage.c:129:_priv_gst_message_initialize: init messages
0:00:00.001932294 143596   0x55702b92c0 INFO                GST_INIT gstcontext.c:85:_priv_gst_context_initialize: init contexts
0:00:00.002509212 143596   0x55702b92c0 INFO      GST_PLUGIN_LOADING gstplugin.c:325:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.002786879 143596   0x55702b92c0 INFO      GST_PLUGIN_LOADING gstplugin.c:233:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.002825379 143596   0x55702b92c0 INFO      GST_PLUGIN_LOADING gstplugin.c:235:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.002870588 143596   0x55702b92c0 INFO            GST_REGISTRY gstregistry.c:1755:ensure_current_registry: reading registry cache: /root/.cache/gstreamer-1.0/registry.aarch64.bin
0:00:00.036123349 143596   0x55702b92c0 INFO            GST_REGISTRY gstregistrybinary.c:666:priv_gst_registry_binary_read_cache: loaded /root/.cache/gstreamer-1.0/registry.aarch64.bin in 0.033184 seconds
0:00:00.036373891 143596   0x55702b92c0 INFO            GST_REGISTRY gstregistry.c:1614:scan_and_update_registry: Validating plugins from registry cache: /root/.cache/gstreamer-1.0/registry.aarch64.bin
0:00:00.039693938 143596   0x55702b92c0 INFO            GST_REGISTRY gstregistry.c:1713:scan_and_update_registry: Registry cache has not changed
0:00:00.039722230 143596   0x55702b92c0 INFO            GST_REGISTRY gstregistry.c:1790:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.039734771 143596   0x55702b92c0 INFO                GST_INIT gst.c:806:init_post: GLib runtime version: 2.66.8
0:00:00.039752563 143596   0x55702b92c0 INFO                GST_INIT gst.c:808:init_post: GLib headers version: 2.66.8
0:00:00.039760146 143596   0x55702b92c0 INFO                GST_INIT gst.c:810:init_post: initialized GStreamer successfully
0:00:00.039833646 143596   0x55702b92c0 INFO            GST_PIPELINE gstparse.c:344:gst_parse_launch_full: parsing pipeline description 'v4l2src device=/dev/video12 ! image/jpeg,width=3840,height=2160,framerate=10/1 ! mppjpegdec format=RGB dma-feature=0 max-errors=-1 ! mpph264enc ! rtph264pay name=pay0 pt=96 ! fpsdisplaysink video-sink=fakesink text-overlay=false '
0:00:00.198072648 143596   0x55702b92c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstvideo4linux2.so" loaded
0:00:00.198424982 143596   0x55702b92c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "v4l2src"
0:00:00.207435621 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseSrc@0x55705401e0> adding pad 'src'
0:00:00.214534799 143596   0x55702b92c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrockchipmpp.so" loaded
0:00:00.214588466 143596   0x55702b92c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "mppjpegdec"
0:00:00.214669258 143596   0x55702b92c0 WARN                     mpp gstmpp.c:121:gst_mpp_use_rga: RGA enabled
0:00:00.215069717 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstVideoDecoder@0x557053f9a0> adding pad 'sink'
0:00:00.215207384 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstVideoDecoder@0x557053f9a0> adding pad 'src'
0:00:00.215501093 143596   0x55702b92c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "mpph264enc"
0:00:00.215925468 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstVideoEncoder@0x5570552910> adding pad 'sink'
0:00:00.215978260 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstVideoEncoder@0x5570552910> adding pad 'src'
0:00:00.220366975 143596   0x55702b92c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrtp.so" loaded
0:00:00.220409267 143596   0x55702b92c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "rtph264pay"
0:00:00.220690726 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstRTPBasePayload@0x55705620a0> adding pad 'src'
0:00:00.220745559 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstRTPBasePayload@0x55705620a0> adding pad 'sink'
0:00:00.225891443 143596   0x55702b92c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstdebugutilsbad.so" loaded
0:00:00.225931984 143596   0x55702b92c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "fpsdisplaysink"
0:00:00.226210235 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstFPSDisplaySink@0x557056a0d0> adding pad 'sink'
0:00:00.226283735 143596   0x55702b92c0 INFO            GST_PIPELINE gstparse.c:344:gst_parse_launch_full: parsing pipeline description 'fakesink'
0:00:00.227565029 143596   0x55702b92c0 INFO      GST_PLUGIN_LOADING gstplugin.c:915:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstcoreelements.so" loaded
0:00:00.227611404 143596   0x55702b92c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "fakesink"
0:00:00.227867487 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseSink@0x55705752b0> adding pad 'sink'
0:00:00.228007196 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:1009:gst_element_get_static_pad: found pad fakesink0:sink
0:00:00.228073404 143596   0x55702b92c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "pipeline"
0:00:00.228338821 143596   0x55702b92c0 INFO            GST_PIPELINE gst/parse/grammar.y:689:gst_parse_perform_link: linking some pad of GstV4l2Src named v4l2src0 to some pad of GstMppJpegDec named mppjpegdec0 (0/0) with caps "image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)10/1"
0:00:00.228377613 143596   0x55702b92c0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "capsfilter"
0:00:00.228561655 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseTransform@0x55705782a0> adding pad 'sink'
0:00:00.228619989 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstelement.c:752:gst_element_add_pad:<GstBaseTransform@0x55705782a0> adding pad 'src'
0:00:00.228664322 143596   0x55702b92c0 INFO              GST_STATES gstbin.c:2115:gst_bin_get_state_func:<pipeline0> getting state
0:00:00.228725864 143596   0x55702b92c0 INFO              GST_STATES gstelement.c:2769:gst_element_continue_state:<capsfilter0> completed state change to NULL
0:00:00.228757655 143596   0x55702b92c0 INFO               GST_EVENT gstevent.c:1610:gst_event_new_reconfigure: creating reconfigure event
0:00:00.228804030 143596   0x55702b92c0 INFO        GST_ELEMENT_PADS gstutils.c:1816:gst_element_link_pads_full: trying to link element v4l2src0:(any) to element capsfilter0:sink
[... cut ...]
0:00:12.216973436 143596   0x5570539860 WARN                     mpp gstmpp.c:251:gst_mpp_rga_do_convert: failed to blit

(gst-launch-1.0:143596): GStreamer-WARNING **: 10:57:02.069: free_priv_data: object finalizing but still has 1 parents (object:0x7fb402e690)

(gst-launch-1.0:143596): GStreamer-CRITICAL **: 10:57:02.069: gst_mini_object_unlock: assertion 'state >= SHARE_ONE' failed
^Chandling interrupt.
Interrupt: Stopping pipeline ...
EOS on shutdown enabled -- Forcing EOS on the pipeline
^C
(base) root@firefly:~# 

```

NIKovachev, 2023-12-20:
the output looks like this:

[image]

bluesky2319, 2023-12-22:
Previously, I had a similar error in Ubuntu, but the issue was fixed after switching to Debian 11. If you use Ubuntu, you can test Debian.

NIKovachev, 2023-12-22:
I'm running stock Debian 11. what could be the problem?

bluesky2319, 2023-12-22:
The problem can be related to RGA. Ensure the RGA driver and library (librga.so, in /usr/lib/aarch64-linux-gnu) are installed in your system. 
https://github.com/airockchip/librga
In my Debian 11 system is like this:
[image]

NIKovachev, 2023-12-24:
I created a new bug for RGA: https://github.com/airockchip/librga/issues/60

there is an option to change the format from RGB to NV12 or similar but the pipeline errors in that case again. 

NOTE: before I hit ^C the pipe hangs
```
root@firefly:~# export GST_MPP_NO_RGA=1
root@firefly:~# gst-launch-1.0 --gst-debug=2 v4l2src device=/dev/video12 ! image/jpeg,width=3840,height=2160,framerate=30/1 ! mppjpegdec format=NV12 dma-feature=1 max-errors=-1 ! mpph265enc ! fpsdisplaysink video-sink="fakesink" text-overlay=false -ve
0:00:00.259724930 239311   0x5566f02180 WARN                     mpp gstmpp.c:126:gst_mpp_use_rga: RGA disabled by env (GST_MPP_NO_RGA)
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)NV12, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)2:4:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstMppH265Enc:mpph265enc0.GstPad:src: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstMppH265Enc:mpph265enc0.GstPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)NV12, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)2:4:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
0:00:00.430072606 239311   0x5566eedb60 WARN          v4l2bufferpool gstv4l2bufferpool.c:814:gst_v4l2_buffer_pool_start:<v4l2src0:pool0:src> Uncertain or not enough buffers, enabling copy threshold
0:00:00.793314357 239311   0x5566eee060 WARN                  mppdec gstmppdec.c:591:gst_mpp_dec_get_frame:<mppjpegdec0> MPP is not able to generate pts
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
[... cut ...]
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason error (-5)
0:00:34.597679046 239311   0x5566eee060 WARN                  mppdec gstmppdec.c:963:gst_mpp_dec_loop:<mppjpegdec0> no matched frame
Execution ended after 0:00:34.318578316
Setting pipeline to NULL ...
Freeing pipeline ...
root@firefly:~# ^C
root@firefly:~# 

```

NIKovachev, 2023-12-29:
hi @JeffyCN ,
how can we solve the last issue `gst_mpp_enc_convert:<mpph265enc0> failed to convert frame`? Do I need to create a new bug for it?

JeffyCN, 2023-12-29:
mpp gstmpp.c:126:gst_mpp_use_rga: RGA disabled by env (GST_MPP_NO_RGA

the rga is disabled by this environment.

i don't know much about rga, if the driver reports error, maybe try to ask the rga maintainer(or committor)

NIKovachev, 2023-12-29:
the last error I pasted here is not related to RGA. Let me create a new bug, this one is for RGA team indeed.

closed 2023-12-29 by NIKovachev

huntlin, 2024-01-05:
> 该问题可能与 RGA 有关。确保系统中已安装 RGA 驱动程序和库（librga.so，位于 /usr/lib/aarch64-linux-gnu 中）。如果已安装，则可以通过更新来解决问题。 https://github.com/airockchip/librga

hello，To update the so library of RGA, you only need to replace the files in the /usr/lib/aarch64-linux-gnu directory?

JeffyCN, 2024-01-05:
normally yes, unless there're some update in headers which requires recompiling.

there are multiple versions of rga drivers and libraries, you can also try other versions of them.

huntlin, 2024-01-05:
> 通常是的，除非标头中有一些更新需要重新编译。
> 
> rga 驱动程序和库有多个版本，您也可以尝试其他版本。
> 
> 或者向维护人员提供您的芯片和驱动程序/库提交号以确认。

At present, my so library can be used normally after local compilation, so I replaced the file in the /usr/lib/aarch64-linux-gnu directory and re-executed the program, but the log is still printed with the old version of the rga_api, how should I modify the version correctly?

JeffyCN, 2024-01-05:
maybe there is another librga somewhere(maybe /usr/lib/?)

you can try to search it in your rootfs and replace them all, or delete the others.

huntlin, 2024-01-05:
> 也许某个地方还有另一个 librga（也许 /usr/lib/？）
> 
> 你可以尝试在你的rootfs中搜索它并替换它们，或者删除其他的。
> 
> 'LD_DEBUG=libs' 可能会告诉您有关加载哪个 librga 的更多信息。

Thank you for your reply, after I replaced it here, it worked fine