JeffyCN/rockchip_mirrors #42: gst_mpp_allocator_import_mppbuf:<mppallocator0> failed to get dmafd

NIKovachev, 2023-12-29:
The below pipeline is running for a minute or so and fails with error `failed to get dmafd`.

```
root@firefly:~# export GST_MPP_NO_RGA=1
root@firefly:~# gst-launch-1.0 --gst-debug=2 v4l2src device=/dev/video12 ! image/jpeg,width=3840,height=2160,framerate=30/1 ! mppjpegdec format=NV12 dma-feature=1 max-errors=-1 ! mpph265enc ! fpsdisplaysink video-sink="fakesink" text-overlay=false -ve
0:00:00.259724930 239311   0x5566f02180 WARN                     mpp gstmpp.c:126:gst_mpp_use_rga: RGA disabled by env (GST_MPP_NO_RGA)
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)NV12, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)2:4:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstMppH265Enc:mpph265enc0.GstPad:src: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstMppH265Enc:mpph265enc0.GstPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)NV12, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)2:4:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
0:00:00.430072606 239311   0x5566eedb60 WARN          v4l2bufferpool gstv4l2bufferpool.c:814:gst_v4l2_buffer_pool_start:<v4l2src0:pool0:src> Uncertain or not enough buffers, enabling copy threshold
0:00:00.793314357 239311   0x5566eee060 WARN                  mppdec gstmppdec.c:591:gst_mpp_dec_get_frame:<mppjpegdec0> MPP is not able to generate pts
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 19, dropped: 0, current: 36.59, average: 36.59
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 35, dropped: 0, current: 30.07, average: 33.29
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 50, dropped: 0, current: 29.99, average: 32.23
.
.
.
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 949, dropped: 0, current: 29.90, average: 30.11
0:00:32.522770313 239311   0x5566eee060 ERROR           mppallocator gstmppallocator.c:166:gst_mpp_allocator_import_mppbuf:<mppallocator0> failed to get dmafd

(gst-launch-1.0:239311): GStreamer-CRITICAL **: 09:47:21.490: gst_mini_object_set_qdata: assertion 'object != NULL' failed
0:00:32.524520613 239311   0x5566eee060 ERROR                 mppenc gstmppenc.c:896:gst_mpp_enc_convert:<mpph265enc0> failed to convert frame
0:00:32.524605488 239311   0x5566eee060 ERROR                 mppenc gstmppenc.c:1149:gst_mpp_enc_handle_frame:<mpph265enc0> not negotiated
0:00:32.524645155 239311   0x5566eee060 WARN                  mppenc gstmppenc.c:1153:gst_mpp_enc_handle_frame:<mpph265enc0> can't handle this frame
0:00:32.559221485 239311   0x5566eee060 ERROR           mppallocator gstmppallocator.c:166:gst_mpp_allocator_import_mppbuf:<mppallocator0> failed to get dmafd

(gst-launch-1.0:239311): GStreamer-CRITICAL **: 09:47:21.526: gst_mini_object_set_qdata: assertion 'object != NULL' failed
0:00:32.561445453 239311   0x5566eee060 ERROR                 mppenc gstmppenc.c:896:gst_mpp_enc_convert:<mpph265enc0> failed to convert frame
0:00:32.561555704 239311   0x5566eee060 ERROR                 mppenc gstmppenc.c:1149:gst_mpp_enc_handle_frame:<mpph265enc0> not negotiated
0:00:32.561575829 239311   0x5566eee060 WARN                  mppenc gstmppenc.c:1153:gst_mpp_enc_handle_frame:<mpph265enc0> can't handle this frame
0:00:32.602139853 239311   0x5566eee060 ERROR           mppallocator gstmppallocator.c:166:gst_mpp_allocator_import_mppbuf:<mppallocator0> failed to get dmafd

(gst-launch-1.0:239311): GStreamer-CRITICAL **: 09:47:21.569: gst_mini_object_set_qdata: assertion 'object != NULL' failed
0:00:32.605662912 239311   0x5566eee060 ERROR                 mppenc gstmppenc.c:896:gst_mpp_enc_convert:<mpph265enc0> failed to convert frame
0:00:32.605855704 239311   0x5566eee060 ERROR                 mppenc gstmppenc.c:1149:gst_mpp_enc_handle_frame:<mpph265enc0> not negotiated
0:00:32.605888371 239311   0x5566eee060 WARN                  mppenc gstmppenc.c:1153:gst_mpp_enc_handle_frame:<mpph265enc0> can't handle this frame
0:00:34.591925602 239311   0x5566eedb60 ERROR                 mppdec gstmppdec.c:1075:gst_mpp_dec_handle_frame:<mppjpegdec0> failed to send packet
0:00:34.591971394 239311   0x5566eedb60 WARN                  mppdec gstmppdec.c:1079:gst_mpp_dec_handle_frame:<mppjpegdec0> can't handle this frame
0:00:34.597338669 239311   0x5566eedb60 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<v4l2src0> error: Internal data stream error.
0:00:34.597369003 239311   0x5566eedb60 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<v4l2src0> error: streaming stopped, reason error (-5)
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason error (-5)
0:00:34.597679046 239311   0x5566eee060 WARN                  mppdec gstmppdec.c:963:gst_mpp_dec_loop:<mppjpegdec0> no matched frame
```

JeffyCN, 2023-12-29:
this usually means the fd is leaking somewhere and reached limit(1024 by default).
try to set "zero-copy-pkt" property of mpp*enc to 0 to copy the encoded pkt to new malloc-ed buffer(without holding dmafd)

NIKovachev, 2023-12-29:
I tried but it didn't work:

```
(base) root@firefly:~# gst-launch-1.0 --gst-debug=2 v4l2src device=/dev/video12 ! image/jpeg,width=3840,height=2160,framerate=30/1 ! mppjpegdec format=NV12 dma-feature=0 max-errors=-1 ! mpph265enc zero-copy-pkt=false ! fpsdisplaysink video-sink="fakesink" text-overlay=false -ve
0:00:00.266725112 870237   0x557819a380 WARN                     mpp gstmpp.c:121:gst_mpp_use_rga: RGA enabled
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)2:4:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstMppH265Enc:mpph265enc0.GstPad:src: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink: caps = video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstMppH265Enc:mpph265enc0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)3840, height=(int)2160, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)2:4:5:1, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = image/jpeg, width=(int)3840, height=(int)2160, framerate=(fraction)30/1, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:5:1
0:00:00.417446608 870237   0x5578182760 WARN          v4l2bufferpool gstv4l2bufferpool.c:814:gst_v4l2_buffer_pool_start:<v4l2src0:pool0:src> Uncertain or not enough buffers, enabling copy threshold
0:00:00.736891856 870237   0x5578182860 WARN                  mppdec gstmppdec.c:591:gst_mpp_dec_get_frame:<mppjpegdec0> MPP is not able to generate pts
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 19, dropped: 0, current: 35.95, average: 35.95
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 35, dropped: 0, current: 30.14, average: 33.04
[... cut ...]
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason error (-5)
0:00:34.590671828 870237   0x5578182860 WARN                  mppdec gstmppdec.c:963:gst_mpp_dec_loop:<mppjpegdec0> no matched frame
Execution ended after 0:00:34.305384458
Setting pipeline to NULL ...
Freeing pipeline ...
(base) root@firefly:~# 

```

JeffyCN, 2023-12-29:
i've uploaded a fix, please try that.

NIKovachev, 2023-12-29:
Issue solved with the latest version! Thank you a lot!

closed 2023-12-29 by NIKovachev