JeffyCN/rockchip_mirrors #46: gstreamer-rockchip: low framerate when encoding under kenel 6.1

amazingfate, 2024-04-14:
mpp kernel driver version: `mpp_service mpp-srv: 6849c1e28b7b author: Ding Wei 2023-12-28 video: rockchip: mpp: px30: fix issue combo_init when device probe`
This issue is introduced by commit https://github.com/JeffyCN/rockchip_mirrors/commit/3e5d9516efa1528b5d31d386f1c462956ae68475

amazingfate, 2024-04-14:
gstreamer test pipeline:
```
gst-launch-1.0 videotestsrc ! queue ! mpph265enc ! fpsdisplaysink video-sink=fakesink -v
```

JeffyCN, 2024-04-14:
the mpp and kernel needs to update maybe, i will try to sync it

JeffyCN, 2024-04-17:
cannot reproduce this issue.
that commit has 2 fix-up in last year, did you test with the newest version?

and i've uploaded the new mpp and kernel  as well.

amazingfate, 2024-04-17:
I've updated mpp and gstreamer-rockchip except kernel to latest and issue is still there.
I will try commits from kernel https://github.com/JeffyCN/mirrors/commits/kernel-6.1/drivers/video/rockchip/mpp to see it that will fix.

JeffyCN, 2024-04-17:
@HermanChen please check it

amazingfate, 2024-04-18:
The latest gstreamer-rockchip fixed. It should be [this commit](https://github.com/JeffyCN/rockchip_mirrors/commit/0aa6bcc5b4c033c990314a69591618f78f46970c) that fixes.

closed 2024-04-18 by amazingfate