JeffyCN/rockchip_mirrors #48: mpph265enc always has ~1 sec latency

NIKovachev, 2024-08-12:
Hello,
mpph265enc always has ~1 sec latency. I tired multiple configurations but I can't lower the latency.  I need a zerolatency option similar to the x265enc or set of parameters that would achieve that. This is real time monitoring application and you have both the view and the camera screen at front of you. 1 sec latency is really noticeable.  Just to mention the network latency is between 5ms and 20ms.

The pipeline I try to tune is very basic:
v4l2src device=/dev/video22 ! mpph265enc ! rtph265pay name=pay0 pt=96

JeffyCN, 2024-08-12:
you can export GST_DEBUG=*mpp*:5 to get logs, and check the handling frame and finishing frame for getting the plugin's letancy.

NIKovachev, 2024-08-12:
hi,
here is the output:
0:00:05.706898785 68096   0x7fa00168c0 DEBUG                    mpp gstmpp.c:255:gst_mpp_rga_do_convert: converted with RGA
0:00:05.768584945 68096   0x7fa00168c0 DEBUG                    mpp gstmpp.c:255:gst_mpp_rga_do_convert: converted with RGA
0:00:05.793923330 68096   0x7fa00168c0 DEBUG                    mpp gstmpp.c:255:gst_mpp_rga_do_convert: converted with RGA
0:00:05.838987819 68096   0x7fa00168c0 DEBUG                    mpp gstmpp.c:255:gst_mpp_rga_do_convert: converted with RGA

I don't have doubt in the mpph265enc performance but it seems tot buffers ~1 sec of frames.

NIKovachev, 2024-08-12:
with GST_DEBUG=mpp:5 that's the only message I get:
 0:00:05.706898785 68096 0x7fa00168c0 DEBUG mpp gstmpp.c:255:gst_mpp_rga_do_convert: converted with RGA

JeffyCN, 2024-08-12:
the basic flow is:
the gst framework feeds buffers to the plugin and trigger 'handling frame' log
the plugin might convert the buffer to aligned dma buf
the plugin feed the buffer to mpp(which is maintained by other people)
the plugin will pull encoded data from mpp in other thread, and then returns it to the gst framework and print 'finishing frame' log.

NIKovachev, 2024-08-12:
ok I got it:
0:00:05.445282397 68294   0x7fb0017cc0 DEBUG                 mppenc gstmppenc.c:1090:gst_mpp_enc_handle_frame:<mpph265enc0> handling frame 99
0:00:05.464952753 68294   0x7fb0015120 DEBUG                 mppenc gstmppenc.c:1034:gst_mpp_enc_poll_packet_locked:<mpph265enc0> finish frame ts=0:00:03.960000000

Is it possible that rtph265pay is buffering then?

JeffyCN, 2024-08-12:
the encoding is async, so that finished frame might not be the input one.

you can check the first handling frame and the first finish frame to get the letancy.

i'm not sure about the rtp plugin, maybe you can try to compare mpph265enc with software h265enc, or maybe enable and check the rtp plugin's log too

NIKovachev, 2024-08-12:
ok thanks for you help! I'll focus on rtph265pay as other people have similar complains as well.

closed 2024-08-12 by NIKovachev