JeffyCN/rockchip_mirrors #50: System Freezing When Running More Than 3 RGA Threads on Orange Pi 5 (RK3588S)

thuongton999, 2024-09-05:
I am developing a GStreamer-based application on the Orange Pi 5 (RK3588) that runs multiple GStreamer pipelines in parallel, each on a different thread. The pipelines are designed to handle video streams from different RTSP sources, utilizing the Rockchip RGA for hardware-accelerated video processing.

When running up to 3 threads, the system functions as expected. However, when I attempt to run more than 3 threads, the system begins to log repeated RGA_BLIT failure messages and eventually freezes, requiring a hard reset. Below is the error message encountered:

```bash
RgaBlit(1485) RGA_BLIT fail: Invalid argument
RgaBlit(1486) RGA_BLIT fail: Invalid argument
handl-fd-vir-phy-hnd-format[0, 209, (nil), (nil), 0, 0]
rect[0, 0, 1920, 1080, 1920, 1088, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
handl-fd-vir-phy-hnd-format[0, 0, 0xfffef00ee9f0, (nil), 0, 0]
rect[0, 0, 640, 480, 640, 480, 512, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
```
Here is my sample GStreamer pipeline:
```
gst-launch-1.0 rtspsrc location=rtsp://<URI> ! queue ! rtph264depay ! h264parse ! mppvideodec ! rgaconvert ! videorate ! video/x-raw,format=RGB,width=640,height=480,framerate=15/1 ! fakesink
```

**Expected Behavior**: The system should be able to handle multiple threads utilizing the RGA without freezing or logging continuous errors.

**Actual Behavior**: When more than 3 threads are run, the system logs repeated RGA_BLIT errors and eventually freezes.

**Environment:**
- Board: Orange Pi 5 (RK3588S)
- OS: Ubuntu Rockchip 22.04

JeffyCN, 2024-09-05:
sorry, don't know much about rga, please ask rga's maintainer(or committer).

JeffyCN, 2024-09-05:
or you can use software conversion(disable rga)

closed 2024-09-05 by JeffyCN