JeffyCN/rockchip_mirrors #51: mpph265enc: AppSink Bitrate changes cause frames to be lost

MentosGuy, 2024-10-20:
Hello,
I recently observed the following:
I use a C application that manages a Gstreamer pipeline as AppSink. Whenever I change the bitrate of the mpph265enc encoder via the AppSink, the encoder loses a few frames that are perceived as microstutters. Is there any way to prevent this? It is a 1080p60 input.

JeffyCN, 2024-10-21:
could you provide a demo to reproduce it?

JeffyCN, 2024-10-27:
try to upload to google drive and send me a link

JeffyCN, 2024-10-28:
i could not reproduce the issue on my rk3588 evb by checking the input frames and output frames.
```
diff --git a/gst/rockchipmpp/gstmppdec.c b/gst/rockchipmpp/gstmppdec.c
index 4f5e447..3f51712 100644
--- a/gst/rockchipmpp/gstmppdec.c
+++ b/gst/rockchipmpp/gstmppdec.c
@@ -26,6 +26,8 @@
 #include "config.h"
 #endif

+#include <stdio.h>
+
 #include "gstmppallocator.h"
 #include "gstmppdec.h"

@@ -991,6 +993,7 @@ gst_mpp_dec_loop (GstVideoDecoder * decoder)
   GST_DEBUG_OBJECT (self, "finish frame ts=%" GST_TIME_FORMAT,
       GST_TIME_ARGS (frame->pts));

+  fprintf(stderr, "finish dec frame %d\n", frame->system_frame_number);
   gst_video_decoder_finish_frame (decoder, frame);

 out:
@@ -1048,6 +1051,8 @@ gst_mpp_dec_handle_frame (GstVideoDecoder * decoder, GstVideoCodecFrame * frame)

   GST_DEBUG_OBJECT (self, "handling frame %d", frame->system_frame_number);

+  fprintf(stderr, "handle dec frame %d\n", frame->system_frame_number);
+
   if (G_UNLIKELY (self->flushing))
     goto flushing;

diff --git a/gst/rockchipmpp/gstmppenc.c b/gst/rockchipmpp/gstmppenc.c
index 65fa180..98f8299 100644
--- a/gst/rockchipmpp/gstmppenc.c
+++ b/gst/rockchipmpp/gstmppenc.c
@@ -26,6 +26,7 @@
 #include "config.h"
 #endif

+#include <stdio.h>
 #include <string.h>

 #include "gstmppallocator.h"
@@ -1067,6 +1068,8 @@ gst_mpp_enc_poll_packet_locked (GstVideoEncoder * encoder)
   GST_DEBUG_OBJECT (self, "finish frame ts=%" GST_TIME_FORMAT,
       GST_TIME_ARGS (frame->pts));

+  fprintf(stderr, "finish enc frame %d\n", frame->system_frame_number);
+
   gst_video_encoder_finish_frame (encoder, frame);

 out:
@@ -1122,6 +1125,8 @@ gst_mpp_enc_handle_frame (GstVideoEncoder * encoder, GstVideoCodecFrame * frame)

   GST_DEBUG_OBJECT (self, "handling frame %d", frame->system_frame_number);

+  fprintf(stderr, "handle enc frame %d\n", frame->system_frame_number);
+
   GST_MPP_ENC_LOCK (encoder);

   if (G_UNLIKELY (self->flushing))
```

there's 3665 frames decoded and 3665 frames encoded.

JeffyCN, 2024-10-28:
i checked your source and output videos on PC, they also have 3665 video packets:
ffprobe -show_packets Output.ts |grep video |wc -l