JeffyCN/rockchip_mirrors #53: no mpp265enc after install gstreamer-rockchip

andrescera, 2024-11-17:
i´d like to ask if there any other requirement in order to get those plugins working, I compiled and installed mpp and gstreamer (1.14)  and then built the packages using:

`meson build -Dprefix=/usr/local -Drockchipmpp=enabled -Drkximage=enabled -Dkmssrc=enabled -Drga=enabled  -Dvpxalphadec=enabled
cd build
ninja
sudo ninja install`

but after running
`gst-inspect-1.0 | grep mpp`
the output I get is 
` rockchipmpp:  mppvideodec: Rockchip's MPP video decoder
rockchipmpp:  mppjpegdec: Rockchip's MPP JPEG image decoder
typefindfunctions: audio/x-musepack: mpc, mpp, mp+
`
is there any other param or requirement I should use in order to get things working properly ?

JeffyCN, 2024-11-17:
it depends on mpp, would not be registered if mpp failed to init

andrescera, 2024-11-17:
thanks it was my kernel version I downgraded from 6.11 to 6.1 and it  worked well

closed 2024-11-17 by andrescera