JeffyCN/rockchip_mirrors #54: Cant find scaling control element under mpph26xenc

FovenX3, 2024-11-21:
Hi, I'm trying to get hardware scaling when using the hardware encoder, I notice that gstreamer-rgaconvert has been replaced by height and width in mpph26xenc.
But after I compile and install rockchip-mpp, under gst-inspect-1.0 mpph264enc, it seems I don't have height and width element.

https://github.com/JeffyCN/rockchip_mirrors/issues/43#issuecomment-1888460687
Thank you!

```
# gst-launch-1.0 --version
gst-launch-1.0 version 1.18.5
GStreamer 1.18.5
http://packages.qa.debian.org/gstreamer1.0
# gst-launch-1.0 v4l2src ! mpph264enc width=1024 height=1024 ! fakesink
WARNING: erroneous pipeline: no property "width" in element "mpph264enc0"
```

```
Factory Details:
  Rank                     primary + 1 (257)
  Long-name                Rockchip Mpp H264 Encoder
  Klass                    Codec/Encoder/Video
  Description              Encode video streams via Rockchip Mpp
  Author                   Randy Li <randy.li@rock-chips.com>, Jeffy Chen <jeffy.chen@rock-chips.com>

Plugin Details:
  Name                     rockchipmpp
  Description              Rockchip Mpp Video Plugin
  Filename                 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrockchipmpp.so
  Version                  1.14.4
  License                  LGPL
  Source module            gst-rockchip
  Binary package           GStreamer Rockchip Plug-ins
  Origin URL               Unknown package origin

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstVideoEncoder
                         +----GstMppEnc
                               +----GstMppH264Enc

Implemented Interfaces:
  GstPreset

Pad Templates:
  SRC template: 'src'
    Availability: Always
    Capabilities:
      video/x-h264
                  width: [ 96, 2147483647 ]
                 height: [ 64, 2147483647 ]
          stream-format: { (string)byte-stream }
              alignment: { (string)au }
                profile: { (string)baseline, (string)main, (string)high }

  SINK template: 'sink'
    Availability: Always
    Capabilities:
      video/x-raw
                 format: { (string)NV12, (string)I420, (string)YUY2, (string)UYVY, (string)BGR16, (string)RGB16, (string)BGR, (string)RGB, (string)ABGR, (string)ARGB, (string)BGRA, (string)RGBA, (string)xBGR, (string)xRGB, (string)BGRx, (string)RGBx, (string)NV12, (string)NV21, (string)I420, (string)YV12, (string)NV16, (string)NV61, (string)BGR16, (string)RGB, (string)BGR, (string)RGBA, (string)BGRA, (string)RGBx, (string)BGRx, (string)NV24, (string)Y444 }
                  width: [ 96, 2147483647 ]
                 height: [ 64, 2147483647 ]

Element has no clocking capabilities.
Element has no URI handling capabilities.

Pads:
  SINK: 'sink'
    Pad Template: 'sink'
  SRC: 'src'
    Pad Template: 'src'

Element Properties:
  arm-afbc            : Input is ARM AFBC compressed format
                        flags: readable, writable
                        Boolean. Default: false
  bps                 : Target BPS (0 = auto calculate)
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 2147483647 Default: 0
  bps-max             : Max BPS (0 = auto calculate)
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 2147483647 Default: 0
  bps-min             : Min BPS (0 = auto calculate)
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 2147483647 Default: 0
  gop                 : Group of pictures starting with I frame (-1 = FPS, 1 = all I frames)
                        flags: readable, writable
                        Integer. Range: -1 - 2147483647 Default: -1
  header-mode         : Header mode
                        flags: readable, writable
                        Enum "MppEncHeaderMode" Default: 0, "first-frame"
                           (0): first-frame      - Only in the first frame
                           (1): each-idr         - In every IDR frames
  level               : H264 level (40~41 = 1080p@30fps, 42 = 1080p60fps, 50~52 = 4K@30fps)
                        flags: readable, writable
                        Enum "GstMppH264Level" Default: 40, "4"
                           (10): 1                - 1
                           (99): 1b               - 1b
                           (11): 1.1              - 1.1
                           (12): 1.2              - 1.2
                           (13): 1.3              - 1.3
                           (20): 2                - 2
                           (21): 2.1              - 2.1
                           (22): 2.2              - 2.2
                           (30): 3                - 3
                           (31): 3.1              - 3.1
                           (32): 3.2              - 3.2
                           (40): 4                - 4
                           (41): 4.1              - 4.1
                           (42): 4.2              - 4.2
                           (50): 5                - 5
                           (51): 5.1              - 5.1
                           (52): 5.2              - 5.2
                           (60): 6                - 6
                           (61): 6.1              - 6.1
                           (62): 6.2              - 6.2
  max-pending         : Max pending frames
                        flags: readable, writable
                        Unsigned Integer. Range: 1 - 16 Default: 16
  max-reenc           : Max re-encode times for one frame
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 3 Default: 1
  min-force-key-unit-interval: Minimum interval between force-keyunit requests in nanoseconds
                        flags: readable, writable
                        Unsigned Integer64. Range: 0 - 18446744073709551615 Default: 0
  name                : The name of the object
                        flags: readable, writable, 0x2000
                        String. Default: "mpph264enc0"
  parent              : The parent of the object
                        flags: readable, writable, 0x2000
                        Object of type "GstObject"
  profile             : H264 profile
                        flags: readable, writable
                        Enum "GstMppH264Profile" Default: 100, "high"
                           (66): baseline         - Baseline
                           (77): main             - Main
                           (100): high             - High
  qos                 : Handle Quality-of-Service events from downstream
                        flags: readable, writable
                        Boolean. Default: false
  qp-delta-ip         : Delta QP between I and P (-1 = default)
                        flags: readable, writable
                        Integer. Range: -1 - 8 Default: -1
  qp-init             : Initial QP (lower value means higher quality)
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 51 Default: 26
  qp-max              : Max QP (0 = default)
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 51 Default: 0
  qp-max-i            : Max Intra QP (0 = default)
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 51 Default: 0
  qp-min              : Min QP (0 = default)
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 51 Default: 0
  qp-min-i            : Min Intra QP (0 = default)
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 51 Default: 0
  rc-mode             : RC mode
                        flags: readable, writable
                        Enum "GstMppEncRcMode" Default: 1, "cbr"
                           (0): vbr              - Variable bitrate
                           (1): cbr              - Constant bitrate
                           (2): fixqp            - Fixed QP
  sei-mode            : SEI mode
                        flags: readable, writable
                        Enum "GstMppEncSeiMode" Default: 0, "disable"
                           (0): disable          - SEI disabled
                           (1): one-seq          - One SEI per sequence
                           (2): one-frame        - One SEI per frame(if changed)
  zero-copy-pkt       : Zero-copy encoded packet
                        flags: readable, writable
                        Boolean. Default: true
(END)
```

JeffyCN, 2024-11-22:
that function is no longer supported(due to rga issues), and disabled by default.

you can try to enable it, but might not work:
install rga dev package, and recompile the gstreamer-rockchip plugin without disabling rga option.
check there's no GST_MPP_NO_RGA environment when running it

FovenX3, 2024-11-22:
Thank you a lot!

closed 2024-11-22 by FovenX3