JeffyCN/rockchip_mirrors #55: How to get mpp wstride and hstride value?

IYJRV, 2024-11-27:
I encountered this issue while developing: the data output by the mppvideodec decoder was aligned. However, during the color conversion, this alignment would affect the display effects (such as appearing with a green border or screen tearing), especially when dealing with H265-encoded videos(It seems that the alignment of H265-encoded video data is more complicated). Therefore, a trimming operation is needed, but trimming requires knowledge of the wstride and hstride. In MPP, the interface for obtaining the horizontal stride is: RK_U32 h_stride = mpp_frame_get_hor_stride(mppFrame); RK_U32 v_stride = mpp_frame_get_ver_stride(mppFrame); However, when using the mppvideodec plugin of gstreamer, it seems that we cannot obtain these two values. I would like to ask if there is any related interface to obtain these two values? Or are there other ways to use RGA to trim the images without these two values?

JeffyCN, 2024-11-27:
the strides and offsets are stores in the gst video info:
https://github.com/JeffyCN/rockchip_mirrors/blob/gstreamer-rockchip/gst/rockchipmpp/gstmpp.h#L49

IYJRV, 2024-11-28:
First, thank you for your reply! When using these macros(GST_MPP_VIDEO_INFO_HSTRIDE, GST_MPP_VIDEO_INFO_VSTRIDE), the values I get seem to be incorrect; For example, for an H.264-encoded video, the aligned width and height should be 1920, 1088 (the buffer size of the decoded data is 3133440，1920 * 1088 * 1.5?), but I still get 1920, 1080 using these macros. There is also a similar problem with H.265 videos. Did I use the wrong macros?

JeffyCN, 2024-11-28:
if the buffer carries video meta, you should upgrade the video info:
https://github.com/JeffyCN/rockchip_mirrors/blob/gstreamer-rockchip/gst/rockchipmpp/gstmppenc.c#L807

IYJRV, 2024-11-28:
It works! Thank you very much for your help!

closed 2024-12-30 by JeffyCN