JeffyCN/rockchip_mirrors #56: Is `drain` unnecessary parameter?

cppcoffee, 2024-12-30:
Is the drain argument to `gst_mpp_enc_reset` not needed? The `self->draining` variable is assigned multiple times:

https://github.com/JeffyCN/rockchip_mirrors/blob/gstreamer-rockchip/gst/rockchipmpp/gstmppenc.c#L459-L468

Also, how can I print out the en_codec elapsed time?

JeffyCN, 2024-12-30:
HACK: The MPP is not capable of handling resets properly. 

it's not needed now, util mpp issue fixed

cppcoffee, 2024-12-30:
I am using the RK3528 to run H.264 encoding and stream it to a local streaming media server, but there is a delay of over 100ms. I would like to test the encoding delay. Is there a way to achieve this?  

Here is the encoding command I am using:  

```bash
gst-launch-1.0 rtspclientsink name=s location=rtsp://localhost:8554/mystream -e v4l2src device=/dev/kvmd-video ! image/jpeg,width=1920,height=1080 ! mppjpegdec ! videoconvert ! mpph264enc rc-mode=vbr ! h264parse ! queue ! s.sink_0
```

JeffyCN, 2024-12-30:
you can try to enable logs(GST_DEBUG env) or add your own logs in those plugins.

closed 2024-12-30 by cppcoffee